SteveMacenski/slam_toolbox #490: Could not find a package configuration file provided by "nav2_map_server"

maksimmasalski, 2022-04-21:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.4 LTS (Focal Fossa)
- Installation type:
  - Cloned slam-toolbox repo, switched to foxy-devel branch
- ROS Version
  - ROS2 foxy

#### Steps to reproduce issue
1. Cloned slam-toolbox repo
2. Run inside of the slam-toolbox directory ` $rosdep install -q -y -r --from-paths src --ignore-src`
3. Then try` $colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug`

#### Expected behavior
Build success

#### Actual behavior
```

maksim@maksim-dev:~/slam_toolbox$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug
Starting >>> slam_toolbox
--- stderr: slam_toolbox                         
CMake Error at CMakeLists.txt:28 (find_package):
  By not providing "Findnav2_map_server.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nav2_map_server", but CMake did not find one.

  Could not find a package configuration file provided by "nav2_map_server"
  with any of the following names:

    nav2_map_serverConfig.cmake
    nav2_map_server-config.cmake

  Add the installation prefix of "nav2_map_server" to CMAKE_PREFIX_PATH or
  set "nav2_map_server_DIR" to a directory containing one of the above files.
  If "nav2_map_server" provides a separate development package or SDK, be
  sure it has been installed.

make: *** [Makefile:3600: cmake_check_build_system] Error 1
---
Failed   <<< slam_toolbox [0.87s, exited with code 2]

Summary: 0 packages finished [0.96s]
  1 package failed: slam_toolbox
  1 package had stderr output: slam_toolbo

```
#### Additional information

SteveMacenski, 2022-04-24:
Do you have nav2 map server installed?

maksimmasalski, 2022-04-25:
Yes, it solved, thank you. @SteveMacenski 
Steve, do you have any examples to enable Slam toolbox with Sick lidar or Rplidar?

closed 2022-04-26 by SteveMacenski