SteveMacenski/slam_toolbox #497: Can't launch localization mode

maxxlife, 2022-06-08:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.4 LTS
- Installation type:
  - from source
- ROS Version
  - ROS1 noetic
- Laser unit:
  - RPLidar A1
  
#### Steps to reproduce issue
I just wanted to run the default localization mode, supplied yaml file with the map files:
#### Expected behavior
I want to see map created by me in the rviz together with the current lidar scannings, it doesn't happen.

#### Actual behavior
```
process[slam_toolbox-2]: started with pid [2907]
[ INFO] [1654720860.101842015]: Node using stack size 40000000
[ INFO] [1654720860.186098291]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1654720860.186687676]: Using plugin solver_plugins::CeresSolver
Load From File /home/maxxlife/Documents/abot_files_maps/maps/june_8/small_room_with_boxes_2_slam_tb_rviz.posegraph 
Mapper <- Module
Mapper <- m_pSequentialScanMatcher
Mapper <- m_pGraph
MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
Mapper <- m_pMapperSensorManager
MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
Mapper <- m_Listeners
**Finished serializing Mapper**
Load From File
**Serializing Dataset**
Dataset <- m_SensorNameLookup
Dataset <- m_Data
Dataset <- m_Lasers
Dataset <- m_pDatasetInfo
**Finished serializing Dataset**
Registering sensor: [Custom Described Lidar]
[ INFO] [1654720860.232167580]: Waiting for incoming scan to get metadata...
[ INFO] [1654720860.391357123]: Got scan!
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0608 22:41:00.392549  2907 problem_impl.cc:635] Parameter block not found: 0x563621c88050. You must add the parameter block to the problem before it can be set constant.
*** Check failure stack trace: ***
[ INFO] [1654720860.404946447]: rviz version 1.14.14
[ INFO] [1654720860.405019643]: compiled against Qt version 5.12.8
[ INFO] [1654720860.405111859]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1654720860.420365515]: Forcing OpenGl version 0.
[slam_toolbox-2] process has died [pid 2907, exit code -6, cmd /home/maxxlife/abot/ros/devel/lib/slam_toolbox/localization_slam_toolbox_node __name:=slam_toolbox __log:=/home/maxxlife/.ros/log/2be4ecbc-e76b-11ec-80d7-855b0e9382c4/slam_toolbox-2.log].
log file: /home/maxxlife/.ros/log/2be4ecbc-e76b-11ec-80d7-855b0e9382c4/slam_toolbox-2*.log
[ INFO] [1654720860.782353531]: Stereo is NOT SUPPORTED
[ INFO] [1654720860.782436351]: OpenGL device: Mesa DRI Intel(R) HD Graphics 4400 (HSW GT2)
[ INFO] [1654720860.782499081]: OpenGl version: 3,0 (GLSL 1,3).
^C[rviz-1] killing on exit
```
#### Additional information

maksimmasalski, 2022-06-08:
> 

I saw in rviz it draw more than 10 circles connected with each other.

maksimmasalski, 2022-06-08:
> Does your localization map have actual nodes? I think that appears when the file didn't contain at least 2 nodes to create a first-constraint.

I saw in rviz it draw more than 10 circles connected with each other. Sorry, it is me) I just logged on another computer now. Because in Ros2 I didn't saw this error, right now checking it again, will let you know if ROS2 has another behavior.

maksimmasalski, 2022-06-08:
@SteveMacenski Do you have some default RVIZ file for `foxy-devel` like in branch `noetic-devel`? I'm using from ros2 branch, but seems it doesn't work. In ros1 branch it worked out of them box:( How to make circles appear in rviz2?

maksimmasalski, 2022-06-08:
@SteveMacenski I tried in ROS2, unfortunately I have ros2 bag with not all topics recorded, but anyway. That is the result:

[localization_slam_toolbox_node-2] Load From File /home/amr_2/Maksim/map_tests/map_rosbag2_2022_05_19-09_00_29.posegraph 
[localization_slam_toolbox_node-2] Mapper <- Module
[localization_slam_toolbox_node-2] Mapper <- m_pSequentialScanMatcher
[localization_slam_toolbox_node-2] Mapper <- m_pGraph
[localization_slam_toolbox_node-2] MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
[localization_slam_toolbox_node-2] MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
[localization_slam_toolbox_node-2] Mapper <- m_pMapperSensorManager
[localization_slam_toolbox_node-2] MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
[localization_slam_toolbox_node-2] Mapper <- m_Listeners
[localization_slam_toolbox_node-2] **Finished serializing Mapper**
[localization_slam_toolbox_node-2] Load From File
[localization_slam_toolbox_node-2] **Serializing Dataset**
[localization_slam_toolbox_node-2] Dataset <- m_SensorNameLookup
[localization_slam_toolbox_node-2] Dataset <- m_Data
[localization_slam_toolbox_node-2] Dataset <- m_Lasers
[localization_slam_toolbox_node-2] Dataset <- m_pDatasetInfo
[localization_slam_toolbox_node-2] **Finished serializing Dataset**
[localization_slam_toolbox_node-2] Registering sensor: [Custom Described Lidar]
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Solver Summary (v 1.14.0-eigen-(3.3.7)-lapack-suitesparse-(5.7.1)-cxsparse-(3.2.0)-eigensparse-openmp-no_tbb)
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2]                                      Original                  Reduced
[localization_slam_toolbox_node-2] Parameter blocks                         5700                     5697
[localization_slam_toolbox_node-2] Parameters                               5700                     5697
[localization_slam_toolbox_node-2] Residual blocks                          3365                     3365
[localization_slam_toolbox_node-2] Residuals                               10095                    10095
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Minimizer                        TRUST_REGION
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Sparse linear algebra library    SUITE_SPARSE
[localization_slam_toolbox_node-2] Trust region strategy     LEVENBERG_MARQUARDT
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2]                                         Given                     Used
[localization_slam_toolbox_node-2] Linear solver          SPARSE_NORMAL_CHOLESKY   SPARSE_NORMAL_CHOLESKY
[localization_slam_toolbox_node-2] Threads                                    50                       50
[localization_slam_toolbox_node-2] Linear solver ordering              AUTOMATIC                     5697
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Cost:
[localization_slam_toolbox_node-2] Initial                          4.040939e+06
[localization_slam_toolbox_node-2] Final                            2.488821e+05
[localization_slam_toolbox_node-2] Change                           3.792057e+06
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Minimizer iterations                        7
[localization_slam_toolbox_node-2] Successful steps                            7
[localization_slam_toolbox_node-2] Unsuccessful steps                          0
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Time (in seconds):
[localization_slam_toolbox_node-2] Preprocessor                         0.002288
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2]   Residual only evaluation           0.003165 (7)
[localization_slam_toolbox_node-2]   Jacobian & residual evaluation     0.014827 (7)
[localization_slam_toolbox_node-2]   Linear solver                      0.040507 (7)
[localization_slam_toolbox_node-2] Minimizer                            0.062390
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Postprocessor                        0.000151
[localization_slam_toolbox_node-2] Total                                0.064830
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Termination:                      CONVERGENCE (Function tolerance reached. |cost_change|/cost: 1.353208e-04 <= 1.000000e-03)
[localization_slam_toolbox_node-2] 
[localization_slam_toolbox_node-2] Registering sensor: [Custom Described Lidar]
[rviz2-1] [INFO] [1654725336.751420208] [rviz2]: Trying to create a map of size 540 x 385 using 1 swatches
[rviz2-1] [ERROR] [1654725336.755370020] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-1] active samplers with a different type refer to the same texture image unit
[localization_slam_toolbox_node-2] [INFO] [1654725337.335127795] [slam_toolbox]: Message Filter dropping message: frame 'sick_scan' at time 1652950274.608 for reason 'Unknown'
[localization_slam_toolbox_node-2] [INFO] [1654725337.879480673] [slam_toolbox]: Message Filter dropping message: frame 'sick_scan' at time 1652950275.048 for reason 'Unknown'
[localization_slam_toolbox_node-2] [INFO] [1654725337.879545480] [slam_toolbox]: Message Filter dropping message: frame 'sick_scan' at time 1652950275.077 for reason 'Unknown'
[... cut ...]
[ERROR] [localization_slam_toolbox_node-2]: process has died [pid 7305, exit code -6, cmd '/home/amr_2/slam_toolbox_ws/install/slam_toolbox/lib/slam_toolbox/localization_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /home/amr_2/slam_toolbox_ws/src/sick_slam_toolbox/config/exp_sick_mapper_params_localization_2.yaml'].
[rviz2-1] [INFO] [1654725343.042492704] [rviz2]: Setting estimate pose: Frame:robot_odom, Position(0.947161, 0.815639, 0), Orientation(0, 0, -0.135057, 0.990838) = Angle: -0.270942

SteveMacenski, 2022-06-08:
It really seems like this should have been resolved as well with recent commits: https://github.com/SteveMacenski/slam_toolbox/commit/2ae75af4c06d7305370a31c44cb225b5a331f2cd but before these errors wouldn't be totally out of the question. I don't see that commit in Noetic so if we found that fixes the issue we can just backport it, but that should only be happening if you're loading bogus map serialized files with no or just 1 node in it. 

That error is totally different from before.

maksimmasalski, 2022-06-09:
> It really seems like this should have been resolved as well with recent commits: [2ae75af](https://github.com/SteveMacenski/slam_toolbox/commit/2ae75af4c06d7305370a31c44cb225b5a331f2cd) but before these errors wouldn't be totally out of the question. I don't see that commit in Noetic so if we found that fixes the issue we can just backport it, but that should only be happening if you're loading bogus map serialized files with no or just 1 node in it.
> 
> That error is totally different from before.

Ok, for the Error in ROS1 I will create one more map, but I remember I had a long pose graph (~10 red dots connected with each other by blue link in rviz)

For the latest Error in ROS2, any ideas why it happens?

maksimmasalski, 2022-06-09:
In ROS2 I used foxy-devel-unfixed branch, will try foxy-devel then

maxxlife, 2022-06-24:
On ROS1 I still can't launch localization mode. 
```
[DEBUG] [1656052464.210353137]: Trying to publish message of type [nav_msgs/OccupancyGrid/3381f2d731d4076ec5c71b0759edbe4e] on a publisher with type [nav_msgs/OccupancyGrid/3381f2d731d4076ec5c71b0759edbe4e]
[DEBUG] [1656052464.210484264]: Trying to publish message of type [nav_msgs/MapMetaData/10cfc8a2818024d3248802c00c95f11b] on a publisher with type [nav_msgs/MapMetaData/10cfc8a2818024d3248802c00c95f11b]
[DEBUG] [1656052464.212198563]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052463.578, count now 1
[DEBUG] [1656052464.336065772]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052463.726, count now 1
[DEBUG] [1656052464.470991599]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052463.861, count now 1
[ INFO] [1656052464.514608300]: Stereo is NOT SUPPORTED
[ INFO] [1656052464.514696851]: OpenGL device: Mesa DRI Intel(R) HD Graphics 4400 (HSW GT2)
[ INFO] [1656052464.514748280]: OpenGl version: 3,0 (GLSL 1,3).
[DEBUG] [1656052464.605818017]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052463.996, count now 1
[DEBUG] [1656052464.752776830]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.131, count now 1
[DEBUG] [1656052464.890399662]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.278, count now 1
[DEBUG] [1656052465.023915487]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.413, count now 1
[DEBUG] [1656052465.170636174]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.549, count now 1
[DEBUG] [1656052465.307075692]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.696, count now 1
[ INFO] [1656052465.322078988]: Creating 1 swatches
[DEBUG] [1656052465.442114433]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.831, count now 1
[DEBUG] [1656052465.578848237]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052464.967, count now 1
[DEBUG] [1656052465.725703582]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052465.101, count now 1
[DEBUG] [1656052465.861079717]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656052465.249, count now 1
[slam_toolbox-2] process has died [pid 44485, exit code -11, cmd /home/maxxlife/abot/ros/devel/lib/slam_toolbox/localization_slam_toolbox_node __name:=slam_toolbox __log:=/home/maxxlife/.ros/log/d160a9c0-f386-11ec-b1a9-6f649275be46/slam_toolbox-2.log].
log file: /home/maxxlife/.ros/log/d160a9c0-f386-11ec-b1a9-6f649275be46/slam_toolbox-2*.log
```
I see it opened RVIZ and loaded map, but lidar scans are frozen and due to the error above everything hangs.
[image]

I know that map has 4 nodes. I tried to apply code fix from the [2ae75af](https://github.com/SteveMacenski/slam_toolbox/commit/2ae75af4c06d7305370a31c44cb225b5a331f2cd) but it didn't solve a problem in ROS1. I still have an error when launch localization mode: 
```
[ INFO] [1656054843.638127234]: Creating 1 swatches
[DEBUG] [1656054843.777136819]: MessageFilter [target=odom ]: Added message in frame abot_lidar at time 1656054843.625, count now 1
[slam_toolbox-2] process has died [pid 50559, exit code -11, cmd /home/maxxlife/abot/ros/devel/lib/slam_toolbox/localization_slam_toolbox_node __name:=slam_toolbox __log:=/home/maxxlife/.ros/log/f7637224-f38b-11ec-b1a9-6f649275be46/slam_toolbox-2.log].
log file: /home/maxxlife/.ros/log/f7637224-f38b-11ec-b1a9-6f649275be46/slam_toolbox-2*.log
```

Anton1B, 2022-09-21:
I may have an idea about this issue, that I've encountered while working on the visual understanding of the pose-graph's behavior with slam_toolbox. 

Required Info:

- Operating System:
Ubuntu 20.04.4 LTS

- Installation type:
from source
- ROS Version
ROS1 noetic

Originally, I started with the commit : add10b465a9aed0a2f7230ba2b103eb6fa8700a0
I could serialize and deserialize maps without any trouble 

But after a while I noticed that the most recent commit was : c5e4d13087a38d0130ddd3caf2a4f87127d0741a
I tested it in a different local branch and noticed that it was working fine for serializing maps. However I could not deserialize maps correctly. 

As I was working on the rviz markers I noticed that the recent additions in [slam_toolbox/src/loop_closure_assistant.cpp](https://github.com/SteveMacenski/slam_toolbox/commit/c5e4d13087a38d0130ddd3caf2a4f87127d0741a#diff-c0327461401c191941c7fdb7f09d22ea24df1727eb135dad90c02d7d32d4d7cf)

And especially :
 
```
 - std::unordered_map<int, Eigen::Vector3d>* graph = solver_->getGraph();

 + karto::MapperGraph * graph = mapper_->GetGraph();
```

was making the localization mode fail. 

I am still a C++ novice but my guess is that in https://github.com/SteveMacenski/slam_toolbox/blob/480646b31542074f85e9c548617b005eafba6c32/slam_toolbox/src/slam_toolbox_common.cpp#L602-L604
slam_toolbox is locking the unique pointer that is necessary to get the graph in the other file. So when we want to deserialize the map we have an error there. If I am wrong, I would be really interested in the real explanation.

SteveMacenski, 2022-09-21:
@hidmic (ROS1) / @malban (ROS2) it looks like PR https://github.com/SteveMacenski/slam_toolbox/pull/467 caused a regression on deserializing files. 

But either way, we can revert that in the meantime if it causes a regression in behavior. This only a visualization improvements.

malban, 2022-09-21:
It sounds like a dead-lock issue?  I'll take a look.

SteveMacenski, 2022-09-22:
@Anton1B can you test https://github.com/SteveMacenski/slam_toolbox/pull/534's change on noetic and let us know if that solves your issue?

Anton1B, 2022-09-23:
In fact, change https://github.com/SteveMacenski/slam_toolbox/pull/534 is already implemented on noetic. But it seems to be ineffective...

https://github.com/SteveMacenski/slam_toolbox/blob/480646b31542074f85e9c548617b005eafba6c32/slam_toolbox/src/slam_toolbox_common.cpp#L206-L215

Anton1B, 2022-09-23:
I tested localization with a map I made IRL and this is what I get 

[image]

[image]

malban, 2022-09-23:
Ok, I can reproduce a crash on noetic.  I should be able to debug what's happening

malban, 2022-09-23:
@Anton1B can you try this MR on noetic? #536

SteveMacenski, 2022-09-23:
https://github.com/SteveMacenski/slam_toolbox/pull/536 resolves!

closed 2022-09-23 by SteveMacenski