SteveMacenski/slam_toolbox #506: localization: recorded map is not visible

maksimmasalski, 2022-06-28:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.4 LTS
- Installation type:
  - from source
- ROS Version
  - ROS2 foxy (foxy-devel branch slam-toolbox)
- Laser unit:
  - RPLidar A3 in ROS2 setup
  
#### Steps to reproduce issue
I'm trying to run localization mode. I created a map, serialized it, and then provided path to map files without their extensions in the config.
When I run localization mode I don't see my existing map in rviz tool. In the log output slam toolbox says that it was able to open posegraph file, no errors.
After launch localization mode I provide estimated pose of the robot using tool 2D Pose Estimate, and it works, but I still don't see my recorded map in rviz. So I need to run robot around to let it create new scannings, which disappear when robot moves far from them, please view screenshots below.
What I'm doing wrong?
**Launched localization mode, running in the room position 1**
[image]
**Moving robot to another position, you see that map started to disappear**
[image]
**Moving robot further, more scannings disappear**
[image]

#### Expected behavior
I expect localization should work like this. I launch this mode, I can see my existing map in rviz, I provide estimated position of the robot on it using rviz, slam toolbox understands it, and puts robot to this position on the existing map. Somehow it works in my case, but I don't see my existing map:(

P.S. Can localization control robot and drive robot to the desired position? For example I provide desired position and localization mode drives robot towards this point?

SteveMacenski, 2022-06-28:
Did you type in the correct & full file path to your serialized map file?

maksimmasalski, 2022-06-29:
> Did you type in the correct & full file path to your serialized map file? Find that serialized file and make sure it contains information and that your name / path is where it is expected to be for loading it.

I typed path to the folder and file (without extensions). For example, in folder "map" I have 4 files: `my_map.pgm`, `my_map.yaml`, `my_map.data`, `my_map.posegraph`. So I provided path like: `/home/maksim/all_maps/map/my_map`. I see that Slam toolbox was able to detect posegraph and open it.

maksimmasalski, 2022-06-29:
I see it loaded map, but then I have an error:
```
[localization_slam_toolbox_node-2] Load From File /home/maksim/Documents/abot_maps/june_28_open_balcon_door/v2/june_28_v2.posegraph 
[localization_slam_toolbox_node-2] Mapper <- Module
[localization_slam_toolbox_node-2] Mapper <- m_pSequentialScanMatcher
[localization_slam_toolbox_node-2] Mapper <- m_pGraph
[localization_slam_toolbox_node-2] MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
[localization_slam_toolbox_node-2] MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
[localization_slam_toolbox_node-2] Mapper <- m_pMapperSensorManager
[localization_slam_toolbox_node-2] MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
[localization_slam_toolbox_node-2] Mapper <- m_Listeners
[localization_slam_toolbox_node-2] **Finished serializing Mapper**
[localization_slam_toolbox_node-2] Load From File
[localization_slam_toolbox_node-2] **Serializing Dataset**
[localization_slam_toolbox_node-2] Dataset <- m_SensorNameLookup
[localization_slam_toolbox_node-2] Dataset <- m_Data
[localization_slam_toolbox_node-2] Dataset <- m_Lasers
[localization_slam_toolbox_node-2] Dataset <- m_pDatasetInfo
[localization_slam_toolbox_node-2] **Finished serializing Dataset**
[localization_slam_toolbox_node-2] Registering sensor: [Custom Described Lidar]
[localization_slam_toolbox_node-2] WARNING: Logging before InitGoogleLogging() is written to STDERR
[localization_slam_toolbox_node-2] F0629 07:51:06.246754  3466 problem_impl.cc:635] Parameter block not found: 0x55fe50ffc480. You must add the parameter block to the problem before it can be set constant.
[localization_slam_toolbox_node-2] *** Check failure stack trace: ***
[ERROR] [localization_slam_toolbox_node-2]: process has died [pid 3466, exit code -6, cmd '/home/maksim/slam_toolbox_ws/install/slam_toolbox/lib/slam_toolbox/localization_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /home/maksim/slam_toolbox_ws/src/sick_slam_toolbox/config/abot_mapper_params_localization.yaml'].
```

maksimmasalski, 2022-06-29:
Okay, so some things that I will add to the docs is a good documentation about Slam toolbox rviz plugin. Instead of providing output path for a serialization after I created map, I provided input path for a .pgm and .yaml file. I will create PR with more detailed information how to use this plugin.

closed 2022-06-29 by maksimmasalski