SteveMacenski/slam_toolbox #507: Localization mode fails

JehanJay, 2022-07-08:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- ROS Version
  - ROS2 Galactic
  
#### Steps to reproduce the issue
I wanted to run the slam toolbox localization mode and basically, I have 4 files (pgm, yaml, posegraph, data) which I generated by the slam toolbox plugin after running in online sync mode. Then I edited the mapper_params_localization.yaml with the corresponding data to start with localization.

#### Actual behavior

> ros2 launch slam_toolbox localization_launch.py
[INFO] [launch]: All log files can be found below /home/jehan/.ros/log/2022-07-08-01-30-10-370697-dell-g15-5511-39324
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [localization_slam_toolbox_node-1]: process started with pid [39326]
[localization_slam_toolbox_node-1] [INFO] [1657236610.520784628] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[localization_slam_toolbox_node-1] [INFO] [1657236610.521728697] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[localization_slam_toolbox_node-1] Load From File /home/jehan/slam_ws/src/debug_slam/ic_2577/results/ex01/ic_2577.posegraph 
[localization_slam_toolbox_node-1] Mapper <- Module
[localization_slam_toolbox_node-1] Mapper <- m_pSequentialScanMatcher
[localization_slam_toolbox_node-1] Mapper <- m_pGraph
[localization_slam_toolbox_node-1] MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
[localization_slam_toolbox_node-1] MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
[localization_slam_toolbox_node-1] Mapper <- m_pMapperSensorManager
[localization_slam_toolbox_node-1] MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
[localization_slam_toolbox_node-1] Mapper <- m_Listeners
[localization_slam_toolbox_node-1] **Finished serializing Mapper**
[localization_slam_toolbox_node-1] Load From File
[localization_slam_toolbox_node-1] **Serializing Dataset**
[localization_slam_toolbox_node-1] Dataset <- m_SensorNameLookup
[localization_slam_toolbox_node-1] Dataset <- m_Data
[localization_slam_toolbox_node-1] Dataset <- m_Lasers
[localization_slam_toolbox_node-1] Dataset <- m_pDatasetInfo
[localization_slam_toolbox_node-1] **Finished serializing Dataset**
[localization_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[ERROR] [localization_slam_toolbox_node-1]: process has died [pid 39326, exit code -11, cmd '/home/jehan/slam_ws/install/slam_toolbox/lib/slam_toolbox/localization_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /home/jehan/slam_ws/install/slam_toolbox/share/slam_toolbox/config/mapper_params_localization.yaml'].

#### Additional information
The config file for the localization -> mapper_params_localization.yaml as follows:

`slam_toolbox:
  ros__parameters:
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_link
    scan_topic: /scan
    mode: localization #mapping

    # if you'd like to start localizing on bringup in a map and pose
    map_file_name: /home/jehan/slam_ws/src/debug_slam/ic_2577/results/ex01/ic_2577
    map_start_pose: [-9.83, -32.4, 0.0]

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.05
    max_laser_range: 20.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 3
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    do_loop_closing: true 
    loop_match_minimum_chain_size: 3
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03
    loop_search_maximum_distance: 3.0

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true
`

So, when I try to load a posegraph file and localize with it, after a while it always fails,  with exact same error. **[ERROR] [localization_slam_toolbox_node-1]: process has died [pid 39326, exit code -11, cmd '/home/jehan/slam_ws/install/slam_toolbox/lib/slam_toolbox/localization_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /home/jehan/slam_ws/install/slam_toolbox/share/slam_toolbox/config/mapper_params_localization.yaml'].**

I sense there's some error in yaml file or the way I launch the localization mode, but I can't quite figure it out, @SteveMacenski  please point out my mistake.

Thanks in advance!

BIRL-xu, 2022-08-04:
I used the default config except **_map_file_name_** and _**_map_start_pose__** and got the same crash.

lucaslins0035, 2023-04-13:
I got the same crash. When I remap `/scan` to another unpublished topic the error does not happen, but of course no localization as well. Maybe something while processing incoming laser data? @SteveMacenski

Huyhust13, 2023-09-27:
I got the same crash in the same ROS version (galactic). 
Is this fixed? 
[image]

Huyhust13, 2023-09-27:
This [commit](https://github.com/SteveMacenski/slam_toolbox/pull/536/commits/38cf601e02c57638784db99135b3be1c92de52da) fixed it, but it wasn't merge to galactic version.

SteveMacenski, 2023-09-27:
Yeah, breaks ABI and Galactic is EOL :/

closed 2024-01-09 by SteveMacenski