SteveMacenski/slam_toolbox #535: Cause of getting -1 (unknown) values on /map/data topic.

MaltheT, 2022-09-22:
**Required Info:**

- Operating System:
  Ubuntu 20.04
- Installation type:
  source
- ROS Version
  ROS2 Foxy
- Version or commit hash:
  Release 2.4.1
- Laser unit:
  RP Lidar A1
  
#### Steps to reproduce issue
Hello,
I want to use slam_toolbox for a project. However, I'm having an issue where slam_toolbox is producing only a single image when I start it. If I echo /map topic I get the following output:
`header:
  stamp:
    sec: 1663833430
    nanosec: 976391541
  frame_id: map
info:
  map_load_time:
    sec: 0
    nanosec: 0
  resolution: 0.05000000074505806
  width: 152
  height: 132
  origin:
    position:
      x: -4.903101768722322
      y: -3.6046353928353208
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
data:
- -1
- -1
- -1
...
- -1
- '...'
---
`
This is what it looks like in rviz2:
[image]

I'm not getting any feedback from the slam_toolbox node as to what is causing the problem. I've been trying to look through the code to see if I could pinpoint what is producing the -1 values, with no luck thus far. 

Any idea why this is happening?

#### Expected behavior
slam_toolbox node continuously generating images of the environment and publishing values in range [0, 100] to /map/data topic.

#### Actual behavior
slam_toolbox node generating only a single image on startup. Afterwards producing only -1 (unknown) values to /map/data topic.

#### Additional information

I'm using robot_localization package to generate odom_1->base_link1 transform. I'm using static TF publishers to generate base_link1->imu1 and base_link1->laser_frame. 

My slam_toolbox configuration file looks like this:
`slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom_1
    map_frame: map
    base_frame: base_link1
    scan_topic: /robotti1/scan
    mode: mapping

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.05 #0.05
    max_laser_range: 12.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true`

And the launch file I'm using to start the slam_toolbox node is the following:

`from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():
    use_sim_time = LaunchConfiguration('use_sim_time')

    declare_use_sim_time_argument = DeclareLaunchArgument(
        'use_sim_time',
        default_value='false',
        description='Use simulation/Gazebo clock')

    start_sync_slam_toolbox_node = Node(
        parameters=[
          get_package_share_directory("drobotti_navigation") + '/config/mapper_params_online_sync.yaml',
          {'use_sim_time': use_sim_time}
        ],
        package='slam_toolbox',
        executable='sync_slam_toolbox_node',
        name='slam_toolbox',
        output='screen')

    ld = LaunchDescription()

    ld.add_action(declare_use_sim_time_argument)
    ld.add_action(start_sync_slam_toolbox_node)

    return ld`

Please let me know if you need more information from me - or if I should move the question to a different forum/location. 

Thanks!

SteveMacenski, 2022-09-22:
http://docs.ros.org/en/noetic/api/nav_msgs/html/msg/OccupancyGrid.html

Please read the message definitions. `-1` is unknown space, because there hasn't been sensor measurements yet to mark that space as obstacles or raytrace them as freespace yet with the given sensor readings. You have to move your robot to new vantage points for new data to be processed into the map.

closed 2022-09-22 by SteveMacenski

MaltheT, 2022-09-23:
Thanks a lot for the fast reply - this solved the issue!