SteveMacenski/slam_toolbox #540: (ref Nav2:#3230) Tb3 is out of bounds of the costmap when navigating and mapping

lmendyk, 2022-10-06:
I have been requested to open the ticket in SLAM repo for the ticket: https://github.com/ros-planning/navigation2/issues/3230 (see the description of the issue from a user perspective)  as it was diagnosed to be related more to SLAM than Navigation2.  

And add details on the installation.
It is the installation based on the binaries for galactic release: 

Package: ros-galactic-slam-toolbox
Version: 2.5.1-1focal.20220730.075909

Package: ros-galactic-navigation2
Version: 1.0.12-1focal.20220730.081413

lmendyk, 2022-10-06:
The launch files are as defined in the tutorial https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html.
This means:
-	ros2 launch turtlebot3_bringup robot.launch.py (at Turtlebot3 Raspberry Pi 4  with LDS-2 -  this is a new model lidar  - it is my first experience no previous version) 

-	ros2 launch nav2_bringup navigation_launch.py use_sim_time:=false ( this “use_sim_time:=false” is my only modification toward the tutorial description)

-	ros2 launch slam_toolbox online_async_launch.py  use_sim_time:=false ( this “use_sim_time:=false” is my only modification toward the tutorial description) 

-	ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/rviz/nav2_default_view.rviz

The configuration is a  default one  which means: 
/opt/ros/galactic/share/slam_toolbox/config/mapper_params_online_async.yaml

[mapper_params_online_async.yaml.txt](https://github.com/SteveMacenski/slam_toolbox/files/9727536/mapper_params_online_async.yaml.txt)

/opt/ros/galactic/share/nav2_bringup/params/nav2_params.yaml
[nav2_params.yaml.txt](https://github.com/SteveMacenski/slam_toolbox/files/9727537/nav2_params.yaml.txt)

As to a bag file I will try to prepare (when I learn how to do it)

lmendyk, 2022-10-06:
Only recently it came to me to try with no nav2 but with teleop, the first test seemed to do the proper mapping but haven't yet finalized the test as my battery went down, so I will need to do further tests. 

As to the simulation, so far I used simulation ether with no SLAM (using the world_map), or the kinetic version with and did not noticed such behavior.

lmendyk, 2022-10-07:
I have done the test with no nav2 (just slam toobox and rviz, the commands to robot sent by a remote controller). 

The result is better (the rooms has been mapped) 
[image]

but finally the problem has happened
  
[image]

lmendyk, 2022-10-07:
I have manged to reproduced the problem while recording the bag:
[tb3_Slam_problem.tar.gz](https://github.com/SteveMacenski/slam_toolbox/files/9735525/tb3_Slam_problem.tar.gz)

The commands which worked for reproducing:
ros2 bag play tb3_3 --clock

ros2 launch slam_toolbox online_async_launch.py use_sim_time:=True

ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args --param use_sim_time:=True

SteveMacenski, 2022-10-07:
```

[async_slam_toolbox_node-1] LaserRangeScan contains 268 range readings, expected 274
[async_slam_toolbox_node-1] LaserRangeScan contains 271 range readings, expected 274
[async_slam_toolbox_node-1] LaserRangeScan contains 271 range readings, expected 274
[async_slam_toolbox_node-1] LaserRangeScan contains 271 range readings, expected 274
```

That's problematic if your lidar sensor isn't producing the correct number of points in a scan, its critical that these are consistent for scan matching and additions to the map. It could be that is causing it, though I'm not confident about that.

Can you replicate this in simulation or with another lidar sensor? I'm trying to isolate this particular sensor / driver to see if this is an issue with the data coming into the system or the system itself.

lmendyk, 2022-10-07:
IN fact it is my first adventure with the real robot - so I don't have access to other robot or other lidar :( .

I don't quite know how to verify if sth is wrong with the odom frame. Do you  mean looking into the bag I recorded and have some algorithm to verify it odom frame does not behave some suddent radical change of values?

Today it took me longer to have this map problem, I even thought that I can't replicate the situation and only after moving into the third room it happned. So to me it soudns as if some numbers has acceded the maximum value :)

I'm not quite sure if I understand the question "What frame is actually "jumping".   the "map" displayed jumps, the robot frames displayed on top of the map also, but I understood it that it is because map->odom->base link  so my undestanding is that it is due to wrong  map->odom done by SLAM.  Do you suggest that this may be a problem of odom->base link?

SteveMacenski, 2022-10-07:
>  Do you suggest that this may be a problem of odom->base link?

Yes, I am wondering about that. In rviz, you should be able to see frames (and you can add labels to them in the side panel) and when the jump happens, what transform is jumping?

lmendyk, 2022-10-07:
I have found the issue which looks being the case here:
https://github.com/ROBOTIS-GIT/turtlebot3/issues/901

SteveMacenski, 2022-10-07:
OK. I'd follow up there then! Glad we found the root cause!

closed 2022-10-07 by SteveMacenski

lmendyk, 2022-10-07:
Thanks for help and ... patience