SteveMacenski/slam_toolbox #559: No pose topic published from slam_toobox

GregoireScf, 2022-12-07:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- ROS Version
  - ROS2 Foxy
- Laser unit:
  - Sick MicroScan3
  
#### Steps to reproduce issue
I'm trying to use slam_toolbox in localization mode to get the pose topic published from the package as it is specified in the documentation.
Basically, after successfully mapped and save my environment, I tried to run all 3 differents launch files in localization mode and everything work super good but the position is not published from slam_toolbox

#### Expected behavior
/pose should be published as geometry_msgs/PoseWithCovarianceStamped message 

#### Actual behavior
no pose topic is published

#### Additional information
I noted that some of my lidar frame are dropped when I launch the nodes but my guess was that my laser scan rate was too high and  that it should not be a problem make things work, right ?

SteveMacenski, 2022-12-07:
Please ask on Ros answers

closed 2022-12-07 by SteveMacenski

PallottaEnrico, 2023-02-06:
Did you find a solution?

GregoireScf, 2023-02-07:
I've explain my workaround [here](https://answers.ros.org/question/409825/slam_toolbox-no-pose-topic-published/)