SteveMacenski/slam_toolbox #564: The slam_toolbox for ROS2 foxy does not publish the topic pose.

BlueBirdHouse, 2023-01-08:
**Required Info:**

- Operating System:
  - Ubuntu 20.04)
- Installation type:
  - binaries
- ROS Version
  - ROS2, foxy
- Version or commit hash:
  - slam_toolbox foxy-devel
  

#### Expected behavior
 Publish the topic pose. 

#### Actual behavior

 Only the topic map is published.

#### Additional information
 I notice that the slam_toolbox for ROS2 foxy does not publish the topic pose ( https://github.com/SteveMacenski/slam_toolbox/tree/foxy-devel ). I have confirmed it on my own virtual machine with the apt-get package. As far as I know, SLAM means mapping and localization. Where does the robot pose goes if the toolbox doesn't publish it? Can I get it from the TF?   
May I use the pose map->Odom->base_link? Is it the same pose published on the pose topic for ROS galactic, humble and rolling?

rsc9421, 2023-02-09:
so there is no topic pose? would be nice to get a pose with covariance

SteveMacenski, 2023-02-09:
see conversation above. There is, but you shouldn’t be using it for that reason. TF is your “topic” with higher bandwidth and accurate current data.

rsc9421, 2023-02-10:
Ok, but the pose that comes from the localization mode (in /tf topic) cannot be fused with other sensors (like GPS) in a EKF because the frames has no covariance information, im correct?