SteveMacenski/slam_toolbox #573: Cannot find package BB

Jiboxiake, 2023-02-15:
System is: 
Operating System: Ubuntu 22.04.1 LTS              
 Kernel: Linux 5.15.0-46-generic
 Architecture: x86-64

TBB is installed from source.
I'm having a similar issue with https://github.com/SteveMacenski/slam_toolbox/issues/469

I simply have a find_package(TBB REQUIRED) and target_link_libraries(tests corelib doctest::doctest parallel_hashmap_lib TBB::tbb) but now I get an error: 
CMake Error at cmake/FindTBB.cmake:397 (file):
  file STRINGS file
  "/home/zhou822/libraries/installed_tbb/include/tbb/tbb_stddef.h" cannot be
  read.

SteveMacenski, 2023-02-15:
This is an issue with your configuration and I unfortunately dont have the resources to debug every users' specific compatibility issues. 

For reference, our binaries turn over fine, so clearly on the 22.04 / ROS distribution of your choice / SLAM Toolbox compiles fine. If you're using a non-standard install of something or modified the software, I'd start looking in those areas.

closed 2023-02-15 by SteveMacenski

Jiboxiake, 2023-02-15:
Thank you. Just wondering about this FindTBB issue. It seems like this should not be run. Do you have any clue what may cause FindTBB to still be run? Thanks.