SteveMacenski/slam_toolbox #574: minimum_time_interval

agross12, 2023-02-16:
I run slam_toolbox in async mode with the following configuration:

slam_toolbox:
  ros__parameters:
    use_sim_time: False
    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_link
    scan_topic: /scan
    mode: mapping
    
    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.2 #if 0 never publishes odometry
    map_update_interval: 2.0
    resolution: 0.05
    max_laser_range: 10.0 #for rastering images
    minimum_time_interval: 1.0
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    enable_interactive_mode: false

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

Expected behaviour:
Scan processed every 1 second, even if the robot is not moving, as per minimum_time_interval parameter.

Actual behaviour:
Scan are not processed.

I've looked through the source code, and I see that minimum_time_interval is used in slam_toolbox_common.cpp to filter range scans.  However later in Mapper.cpp it uses a different variable m_pMinimumTimeInterval which is still at its default value of 3600s.  There does not seem to be any way to set the Mapper variable.  I would expect the default value of this variable to be overriden by the minimum_time_interval parameter as set in the config file.

SteveMacenski, 2023-02-16:
>  Scan processed every 1 second, even if the robot is not moving, as per minimum_time_interval parameter.

That is not how this works, we process scans based on distance travelled and then also throttling by time. If you haven't moved, there's no reason to process a new scan because it will contain no new information. Its actually quite important for good quality results that you only process new scans after you've moved some non-trivial amounts so that there is some overlap, but also some new information.

But we can discuss a change here if there's a strong reason why you want to process on time changes without angle / distance thresholds being met. But the intent of minimum time is to throttle angle/distance, but as a trigger itself.

agross12, 2023-03-06:
The use case I have is that I need to be able to add to the map if a new obstacle is placed near the robot while it isn't moving.  For example if the robot is stopped, and someone walks in front of it, or an object is placed in front of it, I'd expect it to incorporate that information next time it moves.

agross12, 2023-03-06:
I'm also confused why a parameter was created for this if the functionality was not intended to be implemented?

SteveMacenski, 2023-03-06:
I don't think that's a valid application of mapping software, you should have other systems using data to account for the current state of the environment (eg. costmaps or similar). This is a probabilistic method that needs many sensor readings to account of the permanent motion of large objects from different angles.

closed 2023-03-06 by SteveMacenski