SteveMacenski/slam_toolbox #576: [async_slam_toolbox_node-1] [INFO] [1676860532.835088830] [slam_toolbox]: Message Filter dropping message: frame 'laser_frame' at time 1676860532.660 for reason 'discarding message because the queue is full'

feelj, 2023-02-20:
**Required Info:**

- Operating System:
  - 
  - ubuntu 22.04
- Installation type:
  - 
- ROS Version
- ros2 humble
  - 
- Version or commit hash:
  - 
- Laser unit:
  - 
  
#### Steps to reproduce issue

To mapping ydlidar x2, i installed ros2 humble and installed slam_toolbox and ran it. 
"ros2 launch slam_toolbox online_async_launch.py" But i got an error.
[async_slam_toolbox_node-1] [INFO] [1676860532.835088830] [slam_toolbox]: Message Filter dropping message: frame 'laser_frame' at time 1676860532.660 for reason 'discarding message because the queue is full'

#### Expected behavior

#### Actual behavior

#### Additional information

xLeiti, 2023-02-20:
Not an slam_toolbox related issue. Sounds like your tf-tree is not set up properly.  
Slam_tolbox does the map -> odom transform, you need an odom -> base_frame transform and a static transform from base_frame to your laser_frame.
Make sure the odom -> base_frame transform rate keeps up with the laser topic publishing rate

feelj, 2023-02-20:
sry i dont understand 
could you explain for me ? detail?
plz give me the hand :(

xLeiti, 2023-02-20:
Which nodes do you run?
Only ydlidar_ros2_driver and slam_toolbox is not enough, you need an odometry source. You can use the package rf2o_laser_odometry for that if you don't have the required hardware. Install it and adjust the config files.
Here are the adjusted launch commands for following tf-tree: map -> odom -> base_link -> laser_frame

ros2 launch ydlidar_ros2_driver ydlidar_launch.py 
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link laser_frame
ros2 launch rf2o_laser_odometry rf2o_laser_odometry.launch.py laser_scan_topic:=scan
ros2 launch slam_toolbox online_async_launch.py base_frame:=base_link

feelj, 2023-02-21:
Thank you very much, but I tried to execute the code you gave me, but there was an error here.
ros2 launch rf2o_laser_odometry rf2o_laser_odometry.launch.py laser_scan_topic:=scan

xLeiti, 2023-02-21:
This fork works:

```
cd ~/ros2_ws/src
git clone -b humble-devel --single-branch https://github.com/mertsaadet/rf2o_laser_odometry.git
cd ..
colcon build --packages-select rf2o_laser_odometry --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release 
. install/setup.bash
```

feelj, 2023-02-21:
oh my god.. sry.. i run ros2 launch rf2o_laser_odometry rf2o_laser_odometry.launch.py laser_scan_topic:=scan
but error occured
[ERROR] [rf2o_laser_odometry_node-1]: exception occurred while executing process:
Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/execute_local.py", line 546, in __execute_process
    transport, self._subprocess_protocol = await async_execute_process(
  File "/opt/ros/humble/lib/python3.10/site-packages/osrf_pycommon/process_utils/async_execute_process_asyncio/impl.py", line 139, in async_execute_process
    transport, protocol = await _async_execute_process_nopty(
  File "/opt/ros/humble/lib/python3.10/site-packages/osrf_pycommon/process_utils/async_execute_process_asyncio/impl.py", line 45, in _async_execute_process_nopty
    transport, protocol = await loop.subprocess_exec(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1667, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.10/asyncio/unix_events.py", line 207, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.10/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.10/asyncio/unix_events.py", line 799, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1733, in _execute_child
    self._posix_spawn(args, executable, env, restore_signals,
  File "/usr/lib/python3.10/subprocess.py", line 1678, in _posix_spawn
    self.pid = os.posix_spawn(executable, args, env, **kwargs)
OSError: [Errno 8] Exec format error: '/home/car/ros2_ws/install/rf2o_laser_odometry/lib/rf2o_laser_odometry/rf2o_laser_odometry_node'
sorry bro..

SteveMacenski, 2023-02-21:
This is not an issue with SLAM Toolbox, please search and/or ask on ROS Answers.

closed 2023-02-21 by SteveMacenski