SteveMacenski/slam_toolbox #579: A2 lidar always sends 0,0,0 data as distance information, the lidar light(red) is not seen

Dev-Mani-Hdr, 2023-02-25:
**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- ROS Version
  - 
- Version or commit hash:
  - 
- Laser unit:
  - 
  
#### Steps to reproduce issue

#### Expected behavior

#### Actual behavior

#### Additional information

Dev-Mani-Hdr, 2023-02-25:
A2 lidar always sends 0,0,0 data as distance information, the lidar light(red) is not seen.

Initial it worked well for some time around 30hrs, later all the distance information has become 0,0,0  and the lidar light(red) is not glowing.

What could be the issue?

Can we get any service support, we are from Bengalure India.

SteveMacenski, 2023-02-25:
I’m not your lidar’s manufacturer, contact them about hardware failurea

closed 2023-02-25 by SteveMacenski