SteveMacenski/slam_toolbox #582: No map received after ira_laer_tools multi-laser output

Gavriel-CTO, 2023-03-05:
**Required Info:**

- Operating System:
  - Ubunt 20.04
- Installation type:
  - binaries
- ROS Version
  - ROS1 Noetic
- Version or commit hash:
  - Version: 1.5.7-1focal.20230110.192323
- Laser unit:
  - Clearpath Husky Simulator with HUSKY_UST10_ENABLED=1 and HUSKY_UST10_SECONDARY_ENABLED=1
  
#### Steps to reproduce issue
   - Enable both virtual lidars
   - add this to launcher:
   <node pkg="ira_laser_tools" name="laserscan_multi_merger" type="laserscan_multi_merger">
    <param name="destination_frame" value="base_link"/>
    <param name="scan_destination_topic" value="/scan"/>
    <param name="laserscan_topics" value ="/front/scan /rear/scan" />
  </node>

#### Expected behavior
  - I have verified with Rviz and Gmapping that the scans are being combined into the /scan topic and Gmapping will use the scans to build a map.
#### Actual behavior
  - slam_toolbox won't build a map using this method
#### Additional information
  - was hard to find info on the "destination_frame" param, maybe that has something to do with it?

SteveMacenski, 2023-03-06:
That package isn't part of this project - please ask on ROS Answers or look at the data its producing. I can't tell you what the output of that particular package is to know why its not working. We expect a fully formed LaserScan message.

closed 2023-03-06 by SteveMacenski

Gavriel-CTO, 2023-03-08:
The data looks good, not sure that is the issue here.

```
gavriel@gavriel-cto:~$ rostopic info scan
Type: sensor_msgs/LaserScan

Publishers: 
 * /laserscan_multi_merger (http://gavriel-cto:45153/)

Subscribers: 
 * /slam_gmapping (http://gavriel-cto:38545/)
 * /rviz (http://gavriel-cto:34577/)
 * /move_base (http://gavriel-cto:33081/)

gavriel@gavriel-cto:~$ rostopic echo scan
header: 
  seq: 799
  stamp: 
    secs: 30
    nsecs: 604000000
  frame_id: "top_plate_link"
angle_min: -2.359999895095825
angle_max: 2.359999895095825
angle_increment: 0.005799999926239252
time_increment: 0.0
scan_time: 0.03333333134651184
range_min: 0.44999998807907104
range_max: 25.0
ranges: [26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 0.8889434337615967, 0.8728505373001099, 0.8596974015235901, 0.8484830856323242, 0.8399553894996643, 0.8299886584281921, 0.8189652562141418, 0.8121286034584045, 0.8028361201286316, 0.788140594959259, 0.7761656045913696, 26.0, 0.7607190608978271, 0.7575971484184265, 0.7574251890182495, 0.760523796081543, 0.7642761468887329, 0.7662544846534729, 0.7680457830429077, 0.7700027823448181, 0.7723833918571472, 0.7749064564704895, 0.776307225227356, 0.7796602249145508, 0.7844770550727844, 0.7876943945884705, 0.787937343120575, 0.7914657592773438, 0.7960071563720703, 0.7977455258369446, 0.7995938062667847, 0.8057575821876526, 0.8081259727478027, 0.8110975623130798, 0.8128832578659058, 0.8160967826843262, 0.8201541900634766, 0.8233633041381836, 0.8281881809234619, 0.8312384486198425, 0.833400309085846, 0.8355249166488647, 0.8396785855293274, 0.8437479734420776, 0.8449754118919373, 0.8526577949523926, 0.8558814525604248, 0.8587668538093567, 0.863347053527832, 0.8652997016906738, 0.8738859295845032, 0.8787156939506531, 0.878123939037323, 0.883574366569519, 0.8874945044517517, 0.8959623575210571, 0.8984680771827698, 0.902416467666626, 0.9065732955932617, 0.9102147221565247, 0.9170271158218384, 0.9226657152175903, 0.9266858100891113, 0.9311148524284363, 0.9349843263626099, 0.942914605140686, 0.9472222924232483, 0.9536519646644592, 0.9577603340148926, 0.9670928120613098, 0.9724058508872986, 0.9763233661651611, 0.9825613498687744, 0.9861704707145691, 0.9959887862205505, 1.0021259784698486, 1.0079954862594604, 1.0127421617507935, 1.022729754447937, 1.0285512208938599, 1.0342217683792114, 1.0406920909881592, 1.04509437084198, 1.0573290586471558, 1.0670053958892822, 1.071596622467041, 1.0767006874084473, 1.0893694162368774, 1.096360206604004, 1.1027103662490845, 1.110400676727295, 1.1151387691497803, 1.1309458017349243, 1.1380538940429688, 1.1443703174591064, 1.1527687311172485, 1.1695666313171387, 1.174713134765625, 1.184895634651184, 1.192954659461975, 1.2019850015640259, 1.218860149383545, 1.2282124757766724, 1.2372404336929321, 1.2462608814239502, 1.2539896965026855, 1.2750262022018433, 1.2855585813522339, 1.2943938970565796, 1.3028393983840942, 1.3145619630813599, 1.3375927209854126, 1.3470854759216309, 1.3611592054367065, 1.371753454208374, 1.384572982788086, 1.4103001356124878, 1.4205924272537231, 1.4360047578811646, 1.4486364126205444, 1.4611417055130005, 1.490140676498413, 1.505947470664978, 1.5208489894866943, 1.5360971689224243, 1.5504603385925293, 1.567594051361084, 1.5842443704605103, 1.6199878454208374, 1.637354850769043, 1.6557413339614868, 1.6756746768951416, 1.6957688331604004, 1.716138482093811, 1.7363518476486206, 1.7783459424972534, 1.8029897212982178, 1.8265310525894165, 1.8503828048706055, 1.876444935798645, 1.900362491607666, 1.9273090362548828, 1.9532904624938965, 2.013645648956299, 2.0453665256500244, 2.0756068229675293, 2.108285665512085, 2.1431665420532227, 2.1777093410491943, 2.2108874320983887, 2.2519917488098145, 2.291315793991089, 2.3314297199249268, 2.3758554458618164, 2.4196720123291016, 2.513453960418701, 2.5629920959472656, 2.6178934574127197, 2.673447847366333, 2.729774236679077, 2.793353319168091, 2.8571436405181885, 2.9245121479034424, 2.996345043182373, 3.0746309757232666, 3.152695417404175, 3.2363128662109375, 3.3261053562164307, 3.420165777206421, 3.5215048789978027, 3.6311161518096924, 3.745979070663452, 3.870558500289917, 4.000850200653076, 4.146318435668945, 4.297867774963379, 4.462414264678955, 4.643709182739258, 4.845107078552246, 5.057750701904297, 5.295043468475342, 5.555423736572266, 5.845219612121582, 6.1 [... cut ...] , 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0, 26.0]
intensities: []
---

```

SteveMacenski, 2023-03-08:
Your ranges being reported are outside of the range max value.

Gavriel-CTO, 2023-03-09:
Seems they don't handle inf well and just put max+1...

Is the bigger issue that the intensities array is blank? The original data had the array populated with zeros...