SteveMacenski/slam_toolbox #588: Failing build nav2 + slam_toolbox

tommy-erko, 2023-03-24:
**Required Info:**

- Operating System:
  - Ubuntu 22.04 LTS
- Installation type:
  - source
- ROS Version
  - ROS2 rolling
- Version or commit hash:
  - navigation2: origin/main
  - slam_toolbox: origin/ros2
  
#### Steps to reproduce issue
`colcon build` in a workspace with nav2 and slam_toolbox packages

#### Expected behavior
successful build

#### Actual behavior

```
stderr: slam_toolbox
 CMake Error: Files to be generated by multiple different commands: "<path_to_ws>/build/slam_toolbox/ament_cmake_environment_hooks/library_path.dsv"
 CMake Generate step failed.  Build files cannot be regenerated correctly.
 Failed   <<< slam_toolbox [12.6s, exited with code 1]
```

#### Additional information
This setup worked before the latest (2023-03-17) ROS2 rolling synchronization.

tommy-erko, 2023-03-24:
the same happens if I first build nav2 alone (skip nav2_bringup dependency on slam_toolbox) and add and attempt to build slam_toolbox afterwards

SteveMacenski, 2023-03-24:
This is very odd. I'm not sure either. Let me call in backup

clalancette, 2023-03-24:
> This is very odd. I'm not sure either. Let me call in backup CC @clalancette

This is likely due to https://github.com/ament/ament_cmake/pull/416.  There is more information in there, but you'll likely have to do something similar to https://github.com/ros-planning/moveit2/pull/1889

SteveMacenski, 2023-03-24:
Thank you for the note Chris, I'll fix this shortly. That would have taken me much, much longer to try to correlate

SteveMacenski, 2023-03-24:
@tommy-erko thanks for the report! I've fixed it in https://github.com/SteveMacenski/slam_toolbox/pull/589

closed 2023-03-24 by SteveMacenski