SteveMacenski/slam_toolbox #590: Unable to load custom parameter file in SLAM Toolbox on ROS 2 Humble

onurulusoy4, 2023-03-26:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- ROS Version:
  - ROS 2 Humble
- Version or commit hash:
  - ros-humble-slam-toolbox 2.6.4-1jammy.20230302.223844
- Laser unit:
  - libgazebo_ros_ray_sensor.so plugin

#### Steps to reproduce issue
1. Create a custom parameter file `mapper_params_online_async.yaml` 
2. Try to run the SLAM Toolbox using the custom parameter file in the same dir:
```
ros2 launch slam_toolbox online_async_launch.py params_file:=mapper_params_online_async.yaml use_sim_time:=true
```

3. Observe that the SLAM Toolbox does not use the custom parameter file and instead uses the default values (yaml file).

#### Expected behavior
The SLAM Toolbox should use the custom parameter file when provided with the correct path.

#### Actual behavior
The SLAM Toolbox ignores the provided custom parameter file path and uses the default values.

#### Additional information
I have tried multiple methods to pass the custom parameter file path, such as modifying the launch file and using the `ros__parameters` remapping, but none of them worked. The issue persists even when I manually start ROS SLAM with the parameter file path provided.

I have also tried replacing the default parameter file in `/opt/ros/humble/share/slam_toolbox/config/mapper_params_online_async.yaml` with my custom file, but the SLAM Toolbox still seems to use the default values.

Any help or guidance on how to correctly set a custom parameter file for the SLAM Toolbox would be greatly appreciated. Thank you!

SteveMacenski, 2023-03-27:
`params_file:=mapper_params_online_async.yaml` does that path actually exist? 

Also that's the wrong launch configuration if you look at the launch files https://github.com/SteveMacenski/slam_toolbox/blob/ros2/launch/online_async_launch.py#L19

onurulusoy4, 2023-03-28:
Hi, apologies for confusion.
I run bash script below after changing directory to yaml own path.
```
ros2 launch slam_toolbox online_async_launch.py params_file:=mapper_params_online_async.yaml use_sim_time:=true
```

I tried also giving the global path
```
ros2 launch slam_toolbox online_async_launch.py params_file:=/path/to/mapper_params_online_async.yaml use_sim_time:=true
```

But the issue persists, it runs toolbox with a default configuration. 
Then I tried not giving any params and changing the actual config file which is in the path
```
/opt/ros/humble/share/slam_toolbox/config/mapper_params_online_async.yaml
```

And it runs as though I didn't changed anything at all.

Since I couldn't run the toolbox with configuration with bash script, I didn't proceed to fix the problems in my launch file to open it with config in my yaml file.

Thanks for your time.

onurulusoy4, 2023-04-01:
Hi, it was just my my oversight. Since I wrote the part in file mapper_params_online_async.yaml with map like:
```
map_file_name: /path/map.yaml
```
instead of
```
map_file_name: /path/map
```
I found out that when I run the script with --debug option.

Issue has been solved.

closed 2023-04-01 by onurulusoy4