SteveMacenski/slam_toolbox #594: queue size limited when loading a map

stm32f303ret6, 2023-04-19:
**Required Info:**
- Operating System:
Ubuntu 22.04
- Installation type:
binaries
- ROS Version
ROS2 Humble

  
#### Steps to reproduce issue
load this map in localization mode 

#### Expected behavior
i want to load a map of 2500*2500 px, with resolution of 0.01m 
[map_.tar.gz](https://github.com/SteveMacenski/slam_toolbox/files/11275173/map_.tar.gz)

#### Actual behavior
im not able to load the complete map, only a part of the map.
this is the terminal message
[async_slam_toolbox_node-2] [INFO] [1681920085.878850035] [slam_toolbox_localization_with_pose]: Message Filter dropping message: frame 'base_footprint' at time 4932.852 for reason 'discarding message because the queue is full'

#### Additional information
if i do the mapping with a lower resolution lidar, the map has fewer points and i can load it completely. Is there a way to increase the size of queue?

SteveMacenski, 2023-04-19:
> im not able to load the complete map, only a part of the map.

What does that mean? Any pictures or more info? I don't see how that's even possible to only partially load. It either would or wouldn't, I would think. 

Overall, needs far more detail to be able to help.

> [async_slam_toolbox_node-2] [INFO] [1681920085.878850035] [slam_toolbox_localization_with_pose]: Message Filter dropping message: frame 'base_footprint' at time 4932.852 for reason 'discarding message because the queue is full'

That means you don't have TF setup, I'm not going to help debug your URDF / robot state publisher. But that can realistically happen in ROS 2 for the first few seconds after initialization due to discovery traffic. Its not a big deal as long as its only happening when you first start up - in which case its not a problem to worry about.

stm32f303ret6, 2023-04-20:
i want to load this map 
[image]

but i got this:
[image]

Shouldn't the whole map load?

this is my config file:
```yaml
slam_toolbox_localization:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: base_link
    base_frame: base_link
    scan_topic: /mid70_2d
    mode: localization #localization

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    map_file_name: map_
    map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.0 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.05
    max_laser_range: 100.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 3
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    do_loop_closing: true 
    loop_match_minimum_chain_size: 3
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03
    loop_search_maximum_distance: 3.0

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true
```

SteveMacenski, 2023-04-20:
Is that full map actually in the serialized file? It seems to me that its not