SteveMacenski/slam_toolbox #598: LAPACK API not found even though liblapack-dev is installed

sandilyasg, 2023-05-05:
- Operating System:
  - 
- Installation type:
  - 
- ROS Version
  - 
- Version or commit hash:
  - 
- Laser unit:
  - 
  
#### Steps to reproduce issue
I am trying to compile slam_toolbox from source in ROS1 melodic in my catkin workspace. When I enter `catkin_make` I get the following error:
```
CMake Error at /usr/share/cmake-3.10/Modules/FindLAPACK.cmake:361 (message):
  A required library with LAPACK API not found.  Please specify library
  location.
Call Stack (most recent call first):
  slam_toolbox/slam_toolbox/CMakeLists.txt:43 (find_package)
```

#### Expected behavior
I am expecting that it would compile because I have the following packages already installed:
```
libblas-dev is already the newest version (3.7.1-4ubuntu1).
liblapack-dev is already the newest version (3.7.1-4ubuntu1).
libatlas-base-dev is already the newest version (3.10.3-5).
libhdf5-dev is already the newest version (1.10.0-patch1+docs-4)
```
#### Actual behavior

#### Additional information
I have looked at multiple sources online to try and solve this but have been stuck. Here is what I searched so far:
[Error in building source files in Ubuntu #81](https://github.com/dicengine/dice/issues/81)
[Re: [easybuild] CMake not finding LAPACK](https://www.mail-archive.com/easybuild@lists.ugent.be/msg05240.html)
[Install LAPACK and BLAS on Linux-Based Systems](https://coral.ise.lehigh.edu/jild13/2016/07/27/install-lapack-and-blas-on-linux-based-systems/)

sandilyasg, 2023-05-05:
> 

I manually installed the dependencies using `sudo apt install`. When I ran `rosdep install --from-paths src --ignore-src --rosdistro melodic -y` I got:
`#All required rosdeps installed successfully`
but I am still getting the same error:
```
CMake Error at /usr/share/cmake-3.10/Modules/FindLAPACK.cmake:361 (message):
  A required library with LAPACK API not found.  Please specify library
  location.
```
I ran `sudo updatedb` to update the locate database and then ran` locate liblapack.so`. Here is what I got:
```
/etc/alternatives/liblapack.so-x86_64-linux-gnu
/etc/alternatives/liblapack.so.3-x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/liblapack.so
/usr/lib/x86_64-linux-gnu/liblapack.so.3
/usr/lib/x86_64-linux-gnu/atlas/liblapack.so
/usr/lib/x86_64-linux-gnu/atlas/liblapack.so.3
/usr/lib/x86_64-linux-gnu/atlas/liblapack.so.3.10.3
/usr/lib/x86_64-linux-gnu/lapack/liblapack.so
/usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3
/usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3.7.1
/var/lib/dpkg/alternatives/liblapack.so-x86_64-linux-gnu
/var/lib/dpkg/alternatives/liblapack.so.3-x86_64-linux-gnu
```

SteveMacenski, 2023-05-05:
You should consider posting on ROS Answers or similar to see if anyone can help (or google around for similar). Unfortunately, I don't have the resources to fix individual user's IT problems :frowning:. Its working fine on my side and I checked the build farm and it is there as well. If I can't reproduce it, I'm more or less just stabbing aimlessly in the dark.

closed 2023-05-05 by SteveMacenski