SteveMacenski/slam_toolbox #599: cannot change params

muttistefano, 2023-05-25:
Hi, I'm on rolling.
When i launch either the online_async or the online_sync I cannot manage to change the parameters relative the odom frame and scan topic.
when I dump the node parameters I read:

```
/slam_toolbox:
  ros__parameters:
    angle_variance_penalty: '#'
    base_frame: ' '
    ceres_dogleg_type: P
    ceres_linear_solver: a
    ceres_loss_function: r
    ceres_preconditioner: a
    ceres_trust_strategy: m
    coarse_angle_resolution: e
    coarse_search_angle_offset: t
    correlation_search_space_dimension: e
    correlation_search_space_resolution: r
    correlation_search_space_smear_deviation: ' '
    debug_logging: n
    distance_variance_penalty: o
    do_loop_closing: t
    enable_interactive_mode: ' '
    fine_search_angle_offset: s
    interactive_mode: e
    link_match_minimum_response_fine: t

```

I don't get if I'm doing something wrong or else.
Thanks

muttistefano, 2023-05-27:
I noticed that the parameters actually change when I modify them in the config file.
I'm just confused by the fact that when I dump the parameters, I get something unrelated.
Thaks

SteveMacenski, 2023-05-29:
That's probably a CLI artifact, but can't be sure. Either way, sounds like SLAM Toolbox is functioning as expected.

closed 2023-05-29 by SteveMacenski