SteveMacenski/slam_toolbox #607: Cant save a map in Humble

sashag1991, 2023-06-04:
**Required Info:**

- Operating System:
  Ubuntu 22.04) -->

- ROS Version
     ROS2 - humble 
- Version or commit hash:

- Laser unit:
  rplidar
  
#### Steps to reproduce issue
-run: ros2 launch turtlebot4_navigation slam.launch.py
-run: ros2 launch turtlebot4_viz view_robot.launch.py
- save map with the command:  ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'tb4_map'"

#### Expected behavior

create a new map 

#### Actual behavior

The command for creating a new map is not responding 

[image]

sashag1991, 2023-06-05:
I get response : results 255, what does it means ?
I don't see a saved map.

SteveMacenski, 2023-06-05:
Does it work if you use the map server directly? What is the error that map saver gives you (if any)?

> ros2 run nav2_map_server map_saver_cli -f my_map_filepath --ros-args -p map_subscribe_transient_local:=true 

(maybe with and without the transient local bits)

SteveMacenski, 2023-06-05:
It also appears your launch file _doesn't even launch_ the map saver, try ours: https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/slam_launch.py#L85

That could well be your issue. We don't maintain the TB4 code, so you'll have to file a ticket with them.

roni-kreinin, 2023-06-06:
@SteveMacenski From what I can tell, `slam_toolbox` will launch the map saver node when the `/slam_toolbox/save_map` service is called. I just ran it now and had no issues:

```
$ ros2 launch turtlebot4_navigation slam.launch.py

[INFO] [sync_slam_toolbox_node-1]: process started with pid [892896]
[sync_slam_toolbox_node-1] [INFO] [1686073593.372447746] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1686073593.801585870] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1686073593.802337077] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[sync_slam_toolbox_node-1] [INFO] [1686073635.273209568] [slam_toolbox]: SlamToolbox: Saving map as test_map.
[sync_slam_toolbox_node-1] [INFO] [1686073635.872494662] [map_saver]:
[sync_slam_toolbox_node-1]      map_saver lifecycle node launched.
[sync_slam_toolbox_node-1]      Waiting on external lifecycle transitions to activate
[sync_slam_toolbox_node-1]      See https://design.ros2.org/articles/node_lifecycle.html for more information.
[sync_slam_toolbox_node-1] [INFO] [1686073635.872700267] [map_saver]: Creating
[sync_slam_toolbox_node-1] [INFO] [1686073635.880493889] [map_saver]: Configuring
[sync_slam_toolbox_node-1] [INFO] [1686073635.889167639] [map_saver]: Saving map from 'map' topic to 'test_map' file
[sync_slam_toolbox_node-1] [WARN] [1686073635.889259624] [map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[sync_slam_toolbox_node-1] [WARN] [1686073635.889286767] [map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
[sync_slam_toolbox_node-1] [WARN] [map_io]: Image format unspecified. Setting it to: pgm
[sync_slam_toolbox_node-1] [INFO] [map_io]: Received a 446 X 301 map @ 0.05 m/pix
[sync_slam_toolbox_node-1] [INFO] [map_io]: Writing map occupancy data to test_map.pgm
[sync_slam_toolbox_node-1] [INFO] [map_io]: Writing map metadata to test_map.yaml
[sync_slam_toolbox_node-1] [INFO] [map_io]: Map saved
[sync_slam_toolbox_node-1] [INFO] [1686073636.107531218] [map_saver]: Map saved successfully
[sync_slam_toolbox_node-1] [INFO] [1686073636.458686577] [map_saver]: Destroying
```

I believe the issue we are seeing is when there is a namespace applied to the robot. If I namespace it to `robot2` as in #604, I see the same error:

```
$ ros2 launch turtlebot4_navigation slam.launch.py namespace:=robot2

[INFO] [sync_slam_toolbox_node-1]: process started with pid [1171590]
[sync_slam_toolbox_node-1] [INFO] [1686074536.693212747] [robot2.slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1686074537.161467312] [robot2.slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1686074537.161867043] [robot2.slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[sync_slam_toolbox_node-1] [INFO] [1686074555.985937467] [robot2.slam_toolbox]: SlamToolbox: Saving map as test2_map.
[sync_slam_toolbox_node-1] [INFO] [1686074556.661909023] [map_saver]:
[sync_slam_toolbox_node-1]      map_saver lifecycle node launched.
[sync_slam_toolbox_node-1]      Waiting on external lifecycle transitions to activate
[sync_slam_toolbox_node-1]      See https://design.ros2.org/articles/node_lifecycle.html for more information.
[sync_slam_toolbox_node-1] [INFO] [1686074556.662212743] [map_saver]: Creating
[sync_slam_toolbox_node-1] [INFO] [1686074556.670857292] [map_saver]: Configuring
[sync_slam_toolbox_node-1] [INFO] [1686074556.683975197] [map_saver]: Saving map from 'map' topic to 'test2_map' file
[sync_slam_toolbox_node-1] [WARN] [1686074556.684026028] [map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[sync_slam_toolbox_node-1] [WARN] [1686074556.684044407] [map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
[sync_slam_toolbox_node-1] [ERROR] [1686074558.691995497] [map_saver]: Failed to spin map subscription
[sync_slam_toolbox_node-1] [INFO] [1686074558.696973775] [map_saver]: Destroying
[sync_slam_toolbox_node-1] [ros2run]: Process exited with failure 1
```

```
ros2 service call /robot2/slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'test2_map'"
requester: making request: slam_toolbox.srv.SaveMap_Request(name=std_msgs.msg.String(data='test2_map'))

response:
slam_toolbox.srv.SaveMap_Response(result=255)
```

The `map_saver` node is not being namespaced in this instance which is why it fails.

SteveMacenski, 2023-06-06:
We appear to be running the map saver CLI node instead of the main server (I must have wrote this before I made the map saver server) https://github.com/SteveMacenski/slam_toolbox/blob/ros2/src/map_saver.cpp#L64. This was clearly something I wrote in ROS 1 in the era before Nav2 was started 

If you're namespacing then use the map saver CLI directly or use the map saver server's service and specify the topic name. I suppose we could add the topic name parameter here to support namespacing as well.

sashag1991, 2023-06-07:
I think it maybe related to connectivity issue because after robot reboot the map had been saved. Anyway the  command : save map with the command: _ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
data: 'tb4_map'_ not always works from first time

etigafy, 2023-06-13:
@roni-kreinin thanks, that worked for me :)

SteveMacenski, 2023-06-13:
@etigafy @sashag1991 can you test if this works for you? I can backport to Humble https://github.com/SteveMacenski/slam_toolbox/pull/613

MikeDegany, 2023-07-03:
@roni-kreinin Just wanted to let you know that other folks have run into the same problem. I actually had the exact issue when I tried saving the map with the service call command. But, the command you gave did the trick! Although I did get that annoying error message "[map_saver]: Failed to spin map subscription" a bunch of times, I managed to work around it by running the command multiple times.

sashag1991, 2023-07-09:
I think it is a network connection issue. Because after few times it works

SteveMacenski, 2023-08-08:
Can someone test https://github.com/SteveMacenski/slam_toolbox/pull/613 and let me know if it resolves your issues? Then I can merge!