SteveMacenski/slam_toolbox #609: Loop closure repeatbility

67bug, 2023-06-09:
I am trying out slam_toolbox for mapping and am running into issues with loop closure. 
* OS:  Ubuntu 18.04
* ROS: Melodic
* slam-toolbox: 1.1.6-1bionic.20221025.212553
* Lidars: 2 YDLidar TG15s overlapped in their Fielf of view, but each has about a 220-degree FOV, combined with laser matcher
* IMU: yes
* Drive: Differential
* Ceres: Local source install
* Site: a 100m long x 20m "room". There are baggage belts that reduce the ability to "see" the whole room from time to time, but this was addressed with intended loop closures. Lots of reflective walls

I am running `offline.launch`:

```
<launch>

  <arg name="scan_topic_out" default="/scan_merged"/>
  <arg name="destination_frame" default="lidar"/>
  <arg name="scan_topics_in" default="/scan_1 /scan_2"/>

  <param name="/use_sim_time" value="true" />

   <node pkg="ira_laser_tools" name="laserscan_multi_merger" type="laserscan_multi_merger" output="screen">
            <param name="destination_frame" value="$(arg destination_frame)"/>
            <param name="scan_destination_topic" value="$(arg scan_topic_out)"/>
            <param name="laserscan_topics" value ="$(arg scan_topics_in)" />
            <param name="angle_min" value="-3.14159"/>
            <param name="angle_max" value="3.14159"/>
            <param name="angle_increment" value="0.00311202835292"/>
            <param name="scan_time" value="0.1"/>
            <param name="range_min" value="0.05"/>
            <param name="range_max" value="50"/>
	</node>
 
  <!-- <node pkg="tf2_ros" type="static_transform_publisher" name="base_link_to_lidar" args="0.8045 0 0.0417 0 0 0 base_link lidar"/>

  <node pkg="tf2_ros" type="static_transform_publisher" name="base_link_to_odom" args="0 0 0 0 0 0 bbv/odom base_link"/> -->

  <node pkg="slam_toolbox" type="sync_slam_toolbox_node" name="slam_toolbox" output="screen">
    <rosparam command="load" file="<path>/mapper_params_offline.yaml" />
  </node>

</launch>
```

The offline yaml file looks like such:
```
# offline mapping
# Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT #DOGLEG # changing per https://github.com/SteveMacenski/slam_toolbox/issues/198#issuecomment-624895645
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: None # HuberLoss # changing per https://github.com/SteveMacenski/slam_toolbox/issues/198#issuecomment-624895645

# ROS Parameters
odom_frame: bbv/odom
map_frame: map
base_frame: base_link
scan_topic: /scan_merged
mode: mapping #localization
debug_logging: true
throttle_scans: 1
transform_publish_period: 0.02 #if 0 never publishes odometry, default 0.02
map_update_interval: 10.0
resolution: 0.05
max_laser_range: 20.0 #for rastering images
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 14400.
[... 621 lines cut ...]
Preprocessor                           0.0017

  Residual evaluation                  0.0081
  Jacobian evaluation                  0.0198
  Linear solver                        0.0015
Minimizer                              0.0312

Postprocessor                          0.0014
Total                                  0.0344

Termination:                      CONVERGENCE (Parameter tolerance reached. Relative step_norm: 5.405248e-04 <= 1.000000e-03.)

[DEBUG] [1686212403.690646041, 1686209098.576827052]: MessageFilter [target=bbv/odom ]: Added message in frame lidar at time 1686209098.485, count now 0
[DEBUG] [1686212403.886857178, 1686209098.672579104]: MessageFilter [target=bbv/odom ]: Added message in frame lidar at time 1686209098.585, count now 0
[DEBUG] [1686212404.082543589, 1686209098.773478418]: MessageFilter [target=bbv/odom ]: Added message in frame lidar at time 1686209098.684, count now 0
[DEBUG] [1686212404.279028593, 1686209098.869275208]: MessageFilter [target=bbv/odom ]: Added message in frame lidar at time 1686209098.782, count now 0
[DEBUG] [1686212404.476474176, 1686209098.970394927]: MessageFilter [target=bbv/odom ]: Added message in frame lidar at time 1686209098.881, count now 0
[DEBUG] [1686212404.677041920, 1686209099.071367519]: MessageFilter [target=bbv/odom ]: Added message in frame lidar at time 1686209098.979, count now 0
[DEBUG] [1686212404.878615948, 16862090

<more of the same as the graph size increases>
```
So, i know that loop closure is happening the the result is evident in the map generated. In every other run since, i see zero loop closures. 

What would cause this?

SteveMacenski, 2023-06-09:
What is `Waiting for topics ...`? I don't see that in the codebase.

> Today, i tried re-running (as I was debugging something else in my side of the code) and i see no loop closures happening -- same bag file, same settings. And obviously the map is not good at all. What could be causing this?

Was the difference that you were continuing a session or using a bag file vs running on live data? I don't know why you would see an entire run that would not generate loop closures when it did beforehand. I'm trying to understand if anything about your situation is different

67bug, 2023-06-09:
Thanks for the note, @SteveMacenski 

Forgive me for the lack of clarity. The `Waiting for Topics` is a mesage from the `laserscan_multi_merger` node that is combining `scan_1` and `scan_2` to generate `scan_merged`

> Was the difference that you were continuing a session or using a bag file vs running on live data? I don't know why you would see an entire run that would not generate loop closures when it did beforehand. I'm trying to understand if anything about your situation is different.

I run like such:
1. Generate a bag file with scans and odometry and save it with `rosbag record`
2. After the fact, i run `offline.launch` which allows me to play with different parameters for loop closure (with the intent of understanding how these impact the map)
3. Rinse and repeat with different settings in the `mappers_params_offline.yaml`

It is indeed odd that I had loop closures in one run and did not see any in any run since. 

Might I ask this: When running offline, if the processing queue gets large, and of course keeps increasing because the CPU is unable to keep up, will all the messages in the queue eventually get processed? Or will the processing stop when the `rosbag play` stops? Why do i ask this?  

When i run rosbag play without any rate control, the processing queue gets filled and slam_toolbox lags the rosbag play (this is expected). But what i observe is that the bag finishes running and slam_toolbox also stops updating the map which caught me by surprise (may be my observation is not accurate, so wanted to get your view on this)

Hence, I started using the rate option `--rate 0.5` which allow the cpu to catch up to publishes. What i am unsure of is if this rate factor is dropping messages that cause the "close enough" pose to get published to ensure that the loop closure is triggered. I will experiment with this.

Now to address the question listed above about playing with rate control and slam_toolbox stopping processing, I will try using the `--queue` option for rosbag play and the `--keep-alive` option.

Thoughts?

> So perhaps your custom install of Ceres could be to blame if it has a bug or for some reason isn't compatible?

Noted. How would i go about exploring this? As in where do i go hunting?

All this said, this is a great toolbox and FAR more intuitive to use than Cartographer. So, thank you!

SteveMacenski, 2023-06-09:
> When running offline, if the processing queue gets large, and of course keeps increasing because the CPU is unable to keep up, will all the messages in the queue eventually get processed? 

Yes, we store the queue and it gets to them as they can. But I will be honest in saying that I see performance degradation of the map quality if you just throw everything into the system all at once and let it process at max rate.

Part of the reason that I know of is TF2 related. TF2 only keeps so much of the time in its buffers. So If you don't have TF up and providing the relevant data at the relevant times, then obviously things start to degrade by no fault of SLAM Toolbox. Some of the data is just no longer available or might try to use the most recent possible data which could be waaay into the future relative to that scan. 

> But what i observe is that the bag finishes running and slam_toolbox also stops updating the map which caught me by surprise

I don't recall that happening from my personal experience, but there might be a rosbag setting you're missing to not bring things down once complete. There will be data missing though when it stops (TF, maybe others it relies on like the ros clock topic). Sorry, this is a case where I've been off of ROS 1 for so long that my memory is starting to fade about my workflows with respect to it. I'm pretty sure the clock though is part of it. 

> Noted. How would i go about exploring this? As in where do i go hunting?

When you don't see any loop closures, do you still see the Ceres prints?

67bug, 2023-06-09:
> Yes, we store the queue and it gets to them as they can. But I will be honest in saying that I see performance degradation of the map quality if you just throw everything into the system all at once and let it process at max rate.

> Part of the reason that I know of is TF2 related. TF2 only keeps so much of the time in its buffers. So If you don't have TF up and providing the relevant data at the relevant times, then obviously [...]

That has exactly been my finding in that there are definitely artifacts like double walls that originally made me think it was a urdf issue, but when i run it slower and ensure that the queue does not get built up to more than say 10 messages, the "double wall" artifact is gone. So in all candor, it is not slam-toolbox, but the nuances of running large bags fast bag files on a computer that is not able to keep up

> I don't recall that happening from my personal experience, but there might be a rosbag setting you're missing to not bring things down once complete. There will be data missing though when it stops [...]

i have tentatively circumvented this issue by increasing the queue size on rosbag play.  Why that adverb? Because i don't know this to be universally true yet 

> ROS1 vs ROS2

Most of our framework for everything else is built up on ROS1, so migration is going to be a bit tedious; so I am exploring what i can in without changing the foundation just yet. But a migration is on the horizon. 

> When you don't see any loop closures, do you still see the Ceres prints?

No, Ceres prints don't show up at all when there is no loop closure. So i know there is causality there. That's why i was poking at "how do i know it is a ceres issue? Ceres is not even being called, so my working hypothesis is as such (i assume i am entitled to be categorically wrong given my limited understanding of slam-toolbox and the karto-sdk):

1. Running at fractional rates in rosbag play is likely skipping messages needed for loop closure to kick in
2. The same seems to happen when i run rosbag play at a rate of 1, if i don't manage the queue size
3. One way around these two is to increase `loop_search_maximum_distance` from the default 3.0 to a slightly larger value

Based on item 3 above, one follow-up quesiton is how `loop_search_maximum_distance` and `loop_search_space_dimension` are related to each other. Perhaps i should look at the source instead to save you some headspace!

SteveMacenski, 2023-06-09:
Got it, that's why I asked :smile: If you put logs in the Ceres plugin, do you see it being called at all? Just making sure that its not an issue that it is being called, just not doing anything usefully due to invalid inputs, constraints not being reflected, or something else which would point to it not being something in the library core. 

I'm still unclear as to why it works sometimes and not others for you. That doesn't seem sensible. What if you try doing it with only 1 lidar stream instead of merging them? I'm wondering if some of this oddity is related to the scan combiner resulting in badly formatted data on the output which for some reason is harder to correlate correctly. 

> One way around these two is to increase loop_search_maximum_distance from the default 3.0 to a slightly larger value

And that worked? I'd definitely recommend taking a look at the source code. 

> Most of our framework for everything else is built up on ROS1, so migration is going to be a bit tedious; so I am exploring what i can in without changing the foundation just yet. But a migration is on the horizon.

Let me know if I can be of assistance - [ROS 2 Migrations](https://www.opennav.org/consulting) are something Open Navigation offers :smile:

CraftyCranberry, 2024-03-17:
> loop_search_maximum_distance

Hi there! Could you please post your findings?

67bug, 2024-03-19:
@CraftyCranberry I have a bunch of findings that i will capture here, but I have to admit that none of them fit logic as I look at how the code should work. The repo is really designed for large spaces, but there is something strange in my specific implementation that i can't quite figure out precisely. I suspect that either I am seeing a small motor-odom drift or a small imu drift that is making this problem particularly bad for me, or i am dropping messages when i am recoding the rosbag or when playing it back.

But here are my findings:

1. I saw the processing queue getting long and slam_toolbox complaining about this. To address this, i broke the rosbag play duration to smaller chunks to enable the processor to catch up and never have an issue with queuing. This was a promising solution, but is painful to handle especially when the area gets large
1. I noticed that there were times when in multiple runs, the loop closure did not happen at the same instant. For example, I would go from map point A (including pose) to map point B and then come back to map point A [and expect a loop closure], and then head on to map point C (not in the area of A to B zone) and then back to map point A. This loop closure happened almost every time, but did not a few other times -- this defied logic -- so what i did here is drove the robot in such a way that i always came back to the loop closure point A and stayed there for a few seconds with small odom changes to ensure that I had a human-interruptable way to checking if the loop closure occurred or not. I suspect -- i have no way to prove it -- that the message queue is either dropping messages or slam_toolbox is skipping messages without reporting an issue (again, the code does not show a possible way for this to happen). Stopping for a few seconds at the loop closure point ensured that any message drop was not happening. Again, not scientific at all!
1. I found that making multiple runs in a desire to get better loop closure was flawed. In other words, expecting multiple returns to map point B to cause a better loop closure does not always result in a better loop closure. Meaning, that visiting a new map point just once is adequate. It saves time and above all, keeps the 'error' limited in scope. This variation in how I mapped seems to work well and is extremely repeatable.
1. I broke the area into a bunch of smaller regions and ensured that i always started with the same orientation for mapping [to make stitching easier]. So imagine mapping a 1000mx20m room in 100mx20m segments with some overlap and stitching manually offline with an image editor. So for a large area, i tesselate the area into smaller "stitchable" zones and then do an offline stitching of the smaller maps. i know this is something built into slam_toolbox with the continuous mapping, but i wanted an idiot-proof way to make mapping repeatable.

The only other thing halfway reasonable that i can say here, @CraftyCranberry, is that keeping the graphs small-ish helped create great overall small maps and consequently great large maps. This helps me in a few other ways as the individual tesselated maps also change a lot so having to redo a map requires me to just map that one small area of the overall map.