SteveMacenski/slam_toolbox #612: noetic version package requires toolbox_lib which seems to be not existing

sungminner, 2023-06-13:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - I both tried installing package by `sudo apt install ros-noetic-slam-toolbox` and directly building from source code
- ROS Version
  - ROS1 noetic
- Version or commit hash:
  - noetic-devel branch
- Laser unit:
  - Hokuyo
  
#### Steps to reproduce issue
I installed this package via sudo apt install.
In my package where I tried to utilize slam toolbox in, I add a dependency in my CMakeLists.txt like this(line 10):

```
find_package(catkin REQUIRED COMPONENTS
  rospy
  slam_toolbox
)
```

And package.xml like this:
 ```
 <buildtool_depend>catkin</buildtool_depend>
  <build_depend>rospy</build_depend>
  <build_depend>slam_toolbox</build_depend>
  <build_export_depend>rospy</build_export_depend>
  <build_export_depend>slam_toolbox</build_export_depend>
  <exec_depend>rospy</exec_depend>
  <exec_depend>slam_toolbox</exec_depend>
```

And I tried catkin_make. However, it gave me this error(my package name is sd_slam):
```
-- +++ processing catkin package: 'sd_slam'
-- ==> add_subdirectory(sd_slam)
CMake Error at /opt/ros/noetic/share/slam_toolbox/cmake/slam_toolboxConfig.cmake:173 (message):
  Project 'sd_slam' tried to find library 'toolbox_lib'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'slam_toolbox'? Did you find_package() it before the subdirectory
  containing its code is included?
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  sd_slam/CMakeLists.txt:10 (find_package)
```

So I tried looking at slam_toolboxConfig.cmake file, which says:
`set(libraries "ceres_solver_plugin;toolbox_lib;slam_toolbox_rviz_plugin;/usr/lib/x86_64-linux-gnu/libboost_system.so.1.71.0;/usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0;/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0")`

And it includes toolbox_lib.

I also found that keyword on your repo.
It was in slam_toolbox/slam_toolbox/CMakeLists.txt line 65.

```
catkin_package(
    INCLUDE_DIRS
      include
      ${Boost_INCLUDE_DIRS}
      ${EIGEN3_INCLUDE_DIRS}
      ${TBB_INCLUDE_DIRS}
    LIBRARIES
      ceres_solver_plugin
      toolbox_lib                                  <-- HERE
      slam_toolbox_rviz_plugin
```

Any ideas or solution about this issue?

#### Expected behavior

#### Actual behavior

#### Additional information

SteveMacenski, 2023-06-13:
It appears that's and outdated artifact that can be removed. Does that fix your problem? If so, feel free to open a PR and we can make that change

sungminner, 2023-06-14:
I also figured out that the name of the library `slam_toolbox_rviz_plugin` has to be changed into `slam_toolbox_rviz` in slam_toolbox/slam_toolbox/CMakeLists.txt. 

```
catkin_package(
    INCLUDE_DIRS
      include
      ${Boost_INCLUDE_DIRS}
      ${EIGEN3_INCLUDE_DIRS}
      ${TBB_INCLUDE_DIRS}
    LIBRARIES
      ceres_solver_plugin
      slam_toolbox_rviz                                  <-- FIX HERE
    CATKIN_DEPENDS
      message_filters
      nav_msgs
      rosconsole
      roscpp
      sparse_bundle_adjustment
      sensor_msgs
      tf2
      tf
      tf2_ros
      visualization_msgs
      pluginlib
      message_runtime
      tf2_geometry_msgs
      interactive_markers
      slam_toolbox_msgs
    DEPENDS
      EIGEN3
      Boost
)
```

Now it works beautifully! Thanks for your reply, and I'll open a PR!

closed 2023-06-14 by sungminner