SteveMacenski/slam_toolbox #617: Robot jumps around when mapping with YDLIDAR TG30

jie-vs, 2023-06-20:
**Required Info:**

- Operating System:
  - Ubuntu 22.04 (on WSL2) 
- Installation type:
  - from source
- ROS Version
  - ROS2 Humble
- Version or commit hash:
  - branch: ros2
  - commit hash: b08d182553c07cfe3cd2c3aca04493dcd52a60f0
- Laser unit:
  - RPLiDAR S2
  - YDLIDAR TG30
  
#### Steps to reproduce issue
- run `online_async_launch` in mapping mode with this [robot](https://robots.ros.org/megarover-ver.3.0/)
- tested with RPLiDAR S2 and YDLIDAR TG30
- YDLIDAR driver used: https://github.com/YDLIDAR/ydlidar_ros2_driver
- RPLIDAR driver used: https://github.com/Slamtec/sllidar_ros2
- my `mapper_params_online_async.yaml`
  ```yaml
  slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan
    mode: mapping #localization

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    #map_file_name: test_steve
    # map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 3.0
    resolution: 0.05
    max_laser_range: 30.0 # default: 20.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.0
    stack_size_to_use: 40000000 # program needs a larger stack size to serialize large maps
    enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
[... 125 lines cut ...]
[ydlidar_ros2_driver_node-5] [YDLIDAR] Successfully obtained the corrected offset angle[-4.250000] from the lidar[20220105--------]
[ydlidar_ros2_driver_node-5] [YDLIDAR] Lidar init success, Elapsed time 767 ms
[ydlidar_ros2_driver_node-5] [YDLIDAR] Create thread 0x18900640
[ydlidar_ros2_driver_node-5] [YDLIDAR] Successed to start scan mode, Elapsed time 1069 ms
[ydlidar_ros2_driver_node-5] timeout count: 1
[ydlidar_ros2_driver_node-5] [YDLIDAR] Fixed Size: 2020
[ydlidar_ros2_driver_node-5] [YDLIDAR] Sample Rate: 20.00K
[ydlidar_ros2_driver_node-5] [YDLIDAR] Successed to check the lidar, Elapsed time 682 ms
[ydlidar_ros2_driver_node-5] [YDLIDAR] Current Sampling Rate : 20.00K
[ydlidar_ros2_driver_node-5] [YDLIDAR] Now lidar is scanning...
[async_slam_toolbox_node-6] Registering sensor: [Custom Described Lidar]
[rviz2-3] [INFO] [1687230396.726792781] [rviz2]: Trying to create a map of size 253 x 97 using 1 swatches
[rviz2-3] [ERROR] [1687230396.822422349] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result :
[rviz2-3] active samplers with a different type refer to the same texture image unit
[rviz2-3] [INFO] [1687230405.727756620] [rviz2]: Trying to create a map of size 254 x 103 using 1 swatches
[rviz2-3] [INFO] [1687230411.725638515] [rviz2]: Trying to create a map of size 254 x 148 using 1 swatches
[rviz2-3] [INFO] [1687230417.726750586] [rviz2]: Trying to create a map of size 261 x 157 using 1 swatches
```

#### Additional information
- Other than the change in LiDAR, everything else (odom publisher, tf) stays the same.
- Noticed robot jumps more during turns, not much jumping around when moving in a straight line.
- Since no problem occurs when using with RPLIDAR, it seems like its something caused by ydlidar_ros2_driver?
	-  Tried a different TG30, but the same issue occurs
- Could you please explain the potential issues or suggest how to identify them? thanks

SteveMacenski, 2023-06-20:
> Since no problem occurs when using with RPLIDAR, it seems like its something caused by ydlidar_ros2_driver? 

This is not a bug with this code but the YD lidar driver as you mention. Its likely due to incorrectly populating the fields of the LaserScan message (e.g. scan time, angles, etc). I can't debug every sensor manufacturer's drivers.

closed 2023-06-20 by SteveMacenski

jie-vs, 2023-06-21:
Thank you for your reply. I understand that you can't help with debugging the ydlidar's driver. 

I noticed an issue with the data from YDLidar TG30, which looks weird (the value i got from a single point in `/scan/ranges` oscillated between 0 and the measured value). I have already reported it in their [repository](https://github.com/YDLIDAR/ydlidar_ros2_driver/issues/27). 
[image]

This data works when using `gmapping` in ROS 1, but not with `slam_toolbox `in ROS 2. Is there any filter or parameter tweak in `slam_toolbox` that can fix this issue? I appreciate your help.