SteveMacenski/slam_toolbox #619: Does this work with LiDAR only?

scrayish, 2023-06-28:
Hello, I have a quick question regarding data merging in `slam_toolbox_sync`

[image]

I don't quite have a robot, which could give me odometry data, rather I have a handheld device with LiDAR on it (I also have a IMU but by the looks of it, since the diagram doesn't show any signs of using IMU, I guess it's not quite supported). Question is - would this work only with LiDAR and no odometry. I see, they're merged together in PosedScan, and also it's handled like that further on. Is there a hacky way maybe to bypass odometry limitation (i.e. write empty odometry values etc.)?

Thanks in advance.

SteveMacenski, 2023-06-28:
You require odometry published to TF as the documentation shows. How you compute that (wheel encoders, IMU, VIO, lidar odometry, etc) is up to you, but you must provide it.

closed 2023-06-28 by SteveMacenski

ywiyogo, 2023-10-19:
> You require odometry published to TF as the documentation shows. How you compute that (wheel encoders, IMU, VIO, lidar odometry, etc) is up to you, but you must provide it.

@SteveMacenski can slam_toolbox deal with odometry from the wheel encoder only? 
In ROS1, I improved the odometry by combining the IMU through the ros_pose_ekf to have better pose estimation and mapping result. Does the scan matching component in slam_toolbox calculate the best pose estimation so that it can replace the EKF of wheel odometry and IMU?

SteveMacenski, 2023-10-21:
This is SLAM, your state estimation for odometry is still required. You should fuse your odometry sources just like you would in ROS 1. Think of this as a 1:1 replacement for gmapping