SteveMacenski/slam_toolbox #620: Poor Topic Namespacing Practices

agoeckner, 2023-06-29:
By default, the slam_toolbox uses the `/scan`, `/map`, and `/map_metadata` topics: https://github.com/SteveMacenski/slam_toolbox/blob/b08d182553c07cfe3cd2c3aca04493dcd52a60f0/src/slam_toolbox_common.cpp#L144

However, this interferes with ROS 2 namespacing, which is critical for multi-robot operation, by causing all robots to use the same `/map` and `/scan` topics by default.

The topics should instead default to `scan`, `map`, etc. (no leading /) to allow namespaces to be applied.

SteveMacenski, 2023-06-29:
They can still be just as easily remapped, but they’re set in the global namespace so you can namespace slam toolbox without the main topics disconnecting. This is more for the single robot case than the multirobot case. 

Given this is how its been for ~5 years, I’m hesitant to changing it since some may rely on this behavior. You are still able to easily remap these topics at launch time without a problem. Due to a viable work around for your needs, I generally think this should stay this way for now. 

If others comment here and this is an important change for many users, we can revisit.

agoeckner, 2023-06-29:
Sounds reasonable! Thanks.