SteveMacenski/slam_toolbox #621: ROS2  foxy segmentation fault when running async/sync launch files ?

jeremyfix, 2023-07-05:
**Required Info:**

- Operating System:
 Ubuntu 20.04

- Installation type:
Binaries from the ros foxy apt repositories

- ROS Version
ROS2 Foxy

- Version or commit hash:
ros-foxy-slam-toolbox                               2.4.1-1focal.20230606.053256

- Laser unit:
  LDS 01 
  
#### Steps to reproduce issue

I tried starting either with a real robot,a turtlebot, or with a simulated Gazebo environment. 

The simulated environment is ran by the launch turtlebot3_gazebo/launch/turtlebot3_house.launch.py

#### Actual behavior

I always have a segmentation fault , which ever configuration I tried; for example : 

```
➜  TP-CartoTurtlebot git:(master) ✗ ros2 launch slam_toolbox online_sync_launch.py use_sim_time:=true 
[INFO] [launch]: All log files can be found below /home/fix_jer/.ros/log/2023-07-05-11-16-44-539417-stollen-275860
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [275992]
[ERROR] [sync_slam_toolbox_node-1]: process has died [pid 275992, exit code -11, cmd '/opt/ros/foxy/lib/slam_toolbox/sync_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /opt/ros/foxy/share/slam_toolbox/config/mapper_params_online_sync.yaml --params-file /tmp/launch_params_tci16p68'].
```

#### Additional information

I do not know if this is relevant, but I tried running the sync node with gdb , although I do not have the compiled node with debug flags :

```
➜  TP-CartoTurtlebot git:(master) ✗ gdb /opt/ros/foxy/lib/slam_toolbox/sync_slam_toolbox_node
[..]
(gdb) run --ros-args -r __node:=slam_toolbox --params-file /opt/ros/foxy/share/slam_toolbox/config/mapper_params_online_sync.yaml --params-file /tmp/launch_params_ngrih6vn
[...]

Thread 1 "sync_slam_toolb" received signal SIGSEGV, Segmentation fault.                                                                                                                                    
0x00005555555a9efe in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()                                                                                                                   
(gdb) bt                                                                                                                                                                                                   
#0  0x00005555555a9efe in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#1  0x00007ffff7e3f71a in rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >::~SubscriptionOptionsWithAllocator() () from /opt/ros/foxy/lib/libtoolbox_common.so
#2  0x00007ffff7a1bdec in rclcpp::TimeSource::create_clock_sub() () from /opt/ros/foxy/lib/librclcpp.so                                                                                                    #3  0x00007ffff7a1d384 in rclcpp::TimeSource::attachNode(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp
::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_inter
faces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>) () from /opt/ros/foxy/lib/librclcpp.so
#4  0x00007ffff79de227 in rclcpp::node_interfaces::NodeTimeSource::NodeTimeSource(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface
>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shar
ed_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>) () from /opt/ros/foxy/lib/librclcpp.so
#5  0x00007ffff79b86d7 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocato
r<char> > const&, rclcpp::NodeOptions const&) () from /opt/ros/foxy/lib/librclcpp.so
#6  0x00007ffff79b93f8 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) () from /opt/ros/foxy/lib/librclcpp.so
#7  0x00005555555a22e9 in ?? ()
#8  0x00007ffff740b083 in __libc_start_main (main=0x5555555a21c0, argc=8, argv=0x7fffffffd1d8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd1c8)
    at ../csu/libc-start.c:308
#9  0x00005555555a298e in ?? ()
(gdb) quit
A debugging session is active.
```

jeremyfix, 2023-07-05:
A follow-up , instead of installing the version provided by the repositories, I did clone your git in my ros workspace and colcon compiled your package, from the foxy-devel branch ;

now, the localization and mapping are working. 

I do not know if the issue comes from my installation or if there is an issue in the binaries served by the ros repositories for the foxy version.

SteveMacenski, 2023-07-06:
Can't be sure, but that traceback doesn't point to anything in SLAM Toolbox but the rclcpp path only, which is interesting. It might be that there's a bug in foxy, but unfortunately the distribution is EOL so it won't receive any further support :/ 

I think we can close this since source builds work and Foxy is EOL so any issues in Foxy's rclcpp library isn't going to be resolved sine there's no longer support for it as of last month. You could try an uninstall and reinstall in case something got in a funky state. Its hard to say that the issue is slam toolbox when the trace doesn't point to _any_ SLAM Toolbox specific code

closed 2023-07-06 by SteveMacenski