SteveMacenski/slam_toolbox #622: Map not extending properly, new frames appear on top of each other

MaltheT, 2023-07-10:
- Operating System:
  - Ubuntu 20.04 
- Installation type:
  - Source
- ROS Version
  - ROS2 Foxy
- Version or commit hash:
  - 2.4.1-1focal.20230606.053256
- Laser unit:
  - Hokuyo 10LX 
  
#### Steps to reproduce issue
While using the slam_toolbox, I've observed an issue where, instead of the map being extended as I drive straight, new map frames seem to spawn on top of each other. The only configurations that have been modified are in the .yaml file as below:

```
slam_toolbox:
  ros__parameters:
    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_link
    scan_topic: /scan
    mode: mapping #localization
```

I launch slam_toolbox with the command:
```
ros2 launch slam_toolbox online_async_launch.py params_file:=/home/malthe/Repositories/localization_and_mapping/src/slam/config/slam_toolbox.yaml use_sim_time:=false
```

Driving straight for a couple of meters gives me the following output in rviz2:
[image]

#### Expected behavior
As the robot drives straight, the map should continuously extend to reflect the new areas explored by the robot.

#### Actual behavior
Instead of the map being extended, new map frames seem to spawn on top of each other, despite the robot moving in a straight line.

#### Additional information

I have tried to follow various tutorials and guides for setting up slam_toolbox, however, I have not been able to identify the part I'm getting wrong. Any assistance on this would be greatly appreciated. Please let me know if you need more information.

SteveMacenski, 2023-07-10:
This looks to me like you might have an issue with your TF tree where the motion of direction is the opposite of the what it is actually doing. Or, can you give some more context on this test? Maybe there are alot of things moving around that the scan matcher is having an issue with?

MaltheT, 2023-07-11:
Yes indeed you are correct - when I was driving straight in the real world the tf tree was moving backward in rviz. I changed the direction of the input velocity and now it is working perfectly. Thanks a lot!

closed 2023-07-11 by MaltheT