SteveMacenski/slam_toolbox #623: lifelong_slam_toolbox_node does not get build

TAlscher, 2023-07-17:
Hi, 
sorry to bother you. I'm not sure if it's really a "bug" but at least it's a bit misleading.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - apt install
- ROS Version
  - Humble
  
#### Steps to reproduce the issue
- sudo apt install ros-humble-slam-toolbox

- using the "lifelong_slam_toolbox_node" executable in a node 
OR launching it with "ros2 launch slam_toolbox lifelong_launch.py"

#### Expected behavior
lifelong_slam_toolbox_node should be executed with the given parameters

#### Actual behavior
lifelong_slam_toolbox_node' not found on the libexec directory

#### Additional information
This executable is just not part of the executables that are exported/built by the CMakeList. Could be a desired behavior since it's flagged as "highly experimental", but then the lifelong_launch shouldn't be exported as well in my opinion.

SteveMacenski, 2023-07-17:
It is experimental and only available when locally building so you understand its deeply experimental and not ready for production use.

TAlscher, 2023-07-18:
but shouldn't the launch file be excluded then? So at the moment, you are shipping a package with a not working launch file by just adding 
`PATTERN "lifelong_launch.py" EXCLUDE` in the install section?

SteveMacenski, 2023-07-22:
Not a bad idea - will do shortly!