SteveMacenski/slam_toolbox #639: Need help to publish odometry

jesusramondovale, 2023-10-07:
Hi! 

I'm using a custom robot with two linear fixed motors, and a IMU (a basic differential drive)
I'm executing Ubuntu 22.04 with ROS2 Humble on a DEBIX Model-A SBC, and a Portenta board who communicates the commands to the motors. 
I succesfully created the robot's URDF file with all joints and links: 

[image]
[image]

And also I've achieved to start mapping: 
[image]

The problem is, when we move the robot it doesn't know it because nobodys publishing the odometry on /odom topic, nor the odom->base_link transform on /tf 

As I said, we've got a Intel Realsense camera who publishes the IMU values at /camera/camera/imu, and a "/portentatopic" where the current speeds of each motor are published. 

I wonder if anyone could help us with creating a node to publish o /odom and /tf. 

Thank you so much!

SteveMacenski, 2023-10-07:
Checkout first time robot setup guide on navigation.ros.org

closed 2023-10-07 by SteveMacenski

jesusramondovale, 2023-10-07:
I've already checked the set up guide but I don't know what I must do with the IMU and current speeds values in order to get the odometry values published. 

I could also use the diff_drive_controller package but I don't know how to either

SteveMacenski, 2023-10-07:
https://navigation.ros.org/setup_guides/odom/setup_odom.html#robot-localization-demo goes into detail with links to appropriate resources.

jesusramondovale, 2023-10-09:
I'm using ROS2 and the demo example given is using ROS1 libraries.. 

Where can I find a ROS2 demo example? Thank you