SteveMacenski/slam_toolbox #641: SLAM not working on SBC Debix Model-A

jesusramondovale, 2023-10-13:
Hi, I'm using the slma_toolbox package on an SBC DEBIX Model A (2GB RAM) and this specs:

- Microprocessor:
  - i.MX 8M (ARM64)
- Operating System:
  - Ubuntu 22.04 
- Installation type:
  - Binaries (apt-get install)
- ROS Version
  - ROS2 Humble Hawksbill 
- Laser unit:
  - SICK NanoScan3

I succesfully made a map of my room executing all nodes on my DEBIX -except the slam node, which was executed on my Ubuntu Virtual Machine on my x86-64 laptop- but when I try to execute ALL nodes (slam node included) con my DEBIX board, it won't make a map, only just the first "photo" : 

[image]

Do you know whay may be causing this?

SteveMacenski, 2023-10-13:
Nope. Perhaps check htop if your computer is thrashing in the CPU or ran out of memory. I think by your experiment you show that SLAM Toolbox works fine, but its its a hardware related issue, so I’d look there. 

I know I’ve done this on Jetsons and other ARM systems, but I don’t know much about your particular board

jesusramondovale, 2023-10-16:
Hi! I've seen checked it's not a RAM problem (executing all nodes on the SBC only use 1.25GB RAM), but all its 4 cores are +90%. 
May it be something related to CPU usage? 

[image]

jesusramondovale, 2023-10-16:
Tried all modes and no one worked :(

SteveMacenski, 2023-10-16:
I think that htop image speaks for itself. You're out of compute power, you don't have any left to run SLAM unfortunately! Try a more powerful machine. I don't have any exact advise about how to make it work better on your SBC off the top of my head to give you. I'd have to spend some significant time with your particular hardware and situation to look into it and see how we can accelerate or improve things. Unfortunately, that's not something I can do as part of this package's maintainer role. If you're interested in potentially some consulting to help, that might be something I can help with through [Open Navigation](https://www.opennav.org/).

jesusramondovale, 2023-10-17:
Can't I change slam settings or something quickly to make it work?

SteveMacenski, 2023-10-17:
See remark above. 

> I don't have any exact advise about how to make it work better on your SBC off the top of my head to give you.

There are users on Jetsons, RPi4's, etc. I can't know about your particular out of the ordinary SBC without digging very deep into it for you which I can't do without alot more information about your system/product/work. Unfortunately, you're a bit out of luck from a package maintainer's perspective.

jesusramondovale, 2023-10-18:
It was a privlege problem. Odom node must be executed as root in order to communicate with the Portenta control board, so slam node also needs to be executed as root. If not, slam node can't receive the info of the odom node. 
Executed all nodes as root (inside DEBIX SBC) and everything worked fine, even with a 1 CM per pixel resolution. It is clear that it wasn't a CPU problem.

You can close this issue now.