SteveMacenski/slam_toolbox #642: Mapper error

siddux, 2023-10-16:
- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- ROS Version
  - ROS 2 Foxy
- Laser unit:
  - Sick
  
Hi, I'm trying to run slam on my robot (real world, not simulation). It looks like all is running ok but when the robot started to move I get the following error:

```bash
[async_slam_toolbox_node-1] terminate called after throwing an instance of 'std::runtime_error'
[async_slam_toolbox_node-1]   what():  Mapper FATAL ERROR - unable to get pointer in probability search!
[ERROR] [async_slam_toolbox_node-1]: process has died [pid 23469, exit code -6, cmd '/opt/ros/foxy/lib/slam_toolbox/async_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /workspaces/kuka_kmr_ws/install/share/kmr_slam_toolbox/config/KMR_mapper_params_online_async.yaml --params-file /tmp/launch_params_nd0sa422'].
```

Anyone knows where could be the problem? Thanks.

SteveMacenski, 2023-10-16:
Whats your hardware platform? Did you try Humble or newer also to see if its causing issues in newer distributions? 

I've never run into that before, so that's a total new one for me

siddux, 2023-10-16:
I am using [kuka kmr](https://img.directindustry.es/images_di/photo-g/17587-12407502.jpg). I'm not planning to move to Humble as it has a lot of dependencies on other packages only available on Foxy.

I will be working on it tomorrow so if you have any idea about where can comes of the error it will be helpful.

SteveMacenski, 2023-10-16:
What is the computer hardware, I meant.

siddux, 2023-10-16:
Just a regular gigabyte aero laptop. I was running the code under docker, maybe can be related?

neerja123-gupta, 2024-06-21:
Hello, I am having similar issues. I am also using KMR and was trying https://github.com/MortenMDahl/kmriiwa_ws_devel to move the KMR rbase robot and and to create a map, which uses slam_toolbox. I ahving the excat same issue my map node crashes after sometime, whenever I move the robot little bit. Were you able to solve the error? Thanks
This is my inital map 
[image]
and it crashes after sometime giving this error
[image]

Ishansehgal, 2025-10-02:
i am also facing similar issue when i try to tune the parameters. it works fine with the default params but when i add custom parameters it fails.
with these default params everything works:

slam_toolbox:
ros__parameters:
odom_frame: odom
map_frame: map
base_frame: base_link
scan_topic: /scan
use_map_saver: false
mode: mapping

If the below params are commented, it will work, but if I uncomment it and as i move the robot 

[image]

, it will give me the error shown in the image attached to this. 

slam_toolbox:
ros__parameters:
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: None
debug_logging: false
throttle_scans: 1
transform_publish_period: 0.02
map_update_interval: 5.0
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 30.
stack_size_to_use: 400000000
enable_interactive_mode: true
use_scan_matching: true
use_scan_barycenter: true
scan_buffer_maximum_scan_distance: 10.0
link_match_minimum_response_fine: 0.1
link_scan_maximum_distance: 1.5
do_loop_closing: true
loop_match_minimum_chain_size: 10
loop_match_maximum_variance_coarse: 3.0
loop_match_minimum_response_coarse: 0.35
loop_match_minimum_response_fine: 0.45
correlation_search_space_dimension: 0.5
loop_search_space_dimension: 8.0
loop_search_space_resolution: 0.05
loop_search_space_smear_deviation: 0.03
distance_variance_penalty: 0.5
angle_variance_penalty: 1.0
fine_search_angle_offset: 0.00349
coarse_search_angle_offset: 0.349
coarse_angle_resolution: 0.0349
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true
min_pass_through: 2
occupancy_threshold: 0.1
odom_frame: odom
map_frame: map
base_frame: base_link
scan_topic: /scan
use_sim_time: true
mode: mapping
resolution: 0.025
min_laser_range: 0.2
max_laser_range: 20.0
scan_queue_size: 30
minimum_travel_distance: 0.2
minimum_travel_heading: 0.17
correlation_search_space_resolution: 0.02
loop_search_maximum_distance: 15.0
scan_buffer_size: 50
correlation_search_space_smear_deviation: 0.03

can someone help me understand what is wrong with these parameters?