SteveMacenski/slam_toolbox #649: Unable to load library libceres_solver_plugin.so

immanueln98, 2023-11-20:
**Required Info:**

-Operating System:
Ubuntu 22.04
Installation type:
-from source
ROS Version
    ROS2 Humble

Version or commit hash:

    https://github.com/SteveMacenski/slam_toolbox/commit/6f34357ec8dc20dec46f26dbf6837f1baa70ded4
  
#### Steps to reproduce issue
I had installed the slam-toolbox from the apt repository and tried to launch it.

#### Expected behavior

#### Actual behavior
The following terminal output is what resulted when trying to launch it.

`[INFO] [launch]: All log files can be found below /home/imanuel/.ros/log/2023-11-20-08-07-28-986288-imanuel-ROG-Zephyrus-G14-GA401IV-GA401IV-22381
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [22382]
[async_slam_toolbox_node-1] [INFO] [1700460449.143651907] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [FATAL] [1700460449.164812947] [slam_toolbox]: Failed to create solver_plugins::CeresSolver, is it registered and built? Exception: Failed to load library /opt/ros/humble/lib/libceres_solver_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /lib/libceres.so.2: undefined symbol: _ZN6google21kLogSiteUninitializedE, at ./src/shared_library.c:99.
[ERROR] [async_slam_toolbox_node-1]: process has died [pid 22382, exit code 1, cmd '/opt/ros/humble/lib/slam_toolbox/async_slam_toolbox_node --ros-args -r __node:=slam_toolbox --params-file /opt/ros/humble/share/slam_toolbox/config/mapper_params_online_async.yaml --params-file /tmp/launch_params_p_zefd3b'].`

I have tried sudo updating && upgrading my system however it still resulted in the same output. Look forward to your assistance.

SteveMacenski, 2023-11-20:
What OS? Did you install Ceres separately on that computer for other reasons of the non-rosdistro version?

immanueln98, 2023-11-20:
I am on Ubuntu 22.04 and I initially did not install ceres separately. I then tried doing so and it still resulted in the same error.

SteveMacenski, 2023-11-20:
I'm not sure, I can't reproduce this on my side. That makes me think its something to do with your system's configuration and likely installed some version of Ceres or Glogging that was incompatible. A quick google search on the symbol seems to suggest the same thing: https://github.com/facebook/watchman/issues/1066

Are you having this issue with the binaries?

immanueln98, 2023-11-21:
What ceres and Glog version is compatible with the latest version of SLAM if I may ask? I had initially only installed Ubuntu 22.04 with ROS2 humble without installing ceres and glog separately.

which binaries are you referring to if I may ask? its failing to load the library /lib/libceres.so with dlopen.

I went through the issue you referred me to, and I was wondering if libgoogle-glog0v5 package is located in the right place based of the terminal output below:
[image]

SteveMacenski, 2023-11-21:
The version in the apt server https://github.com/ros/rosdistro/blob/5fcfe66f79761c162b90bb78090b46ddaf58cbf4/rosdep/base.yaml#L2970-L2979

I tested by compiling from source in 22.04 and I don't have any issues - nor does the binaries built in the build farm or the binaries I have in Humble as well. I can't speak to your particular issue, but its not related to this codebase given multiple reproducable examples and no other user reports with hundreds of users on Humble / 22.04.

immanueln98, 2023-12-13:
Hi Steve so I figured out why I am having this issue. 
I had upgraded my machine from Ubuntu 20 to 22 initially. I then did a fresh install of Ubuntu 22.04 and everything ran as 
expected! 
So I guess one has to avoid upgrading Ubuntu in order to run this package.

SteveMacenski, 2023-12-13:
That is incorrect, almost everyone on humble, rolling, iron, etc. use 22.04.