SteveMacenski/slam_toolbox #651: Add support to ROS 2 Components

AntoBrandi, 2023-11-21:
#### Feature description
As all the [Nav2 nodes](https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/navigation_launch.py#L218) support [Composition](https://docs.ros.org/en/humble/index.html) we might really benefit from having the `slam_toolbox` nodes supporting composition as well

#### Implementation considerations
Composable Nodes might look like [rtabmap_ros](https://github.com/introlab/rtabmap_ros/tree/ros2#rtabmap_ros)

```
~$ ros2 component types
rtabmap_slam
  rtabmap_slam::CoreWrapper
rtabmap_sync
  rtabmap_sync::RGBDSync
  rtabmap_sync::StereoSync
  rtabmap_sync::RGBSync
  rtabmap_sync::RGBDXSync
rtabmap_odom
  rtabmap_odom::RGBDOdometry
  rtabmap_odom::StereoOdometry
  rtabmap_odom::ICPOdometry
rtabmap_util
  rtabmap_util::RGBDRelay
  rtabmap_util::RGBDSplit
  rtabmap_util::LidarDeskewing
  rtabmap_util::PointCloudXYZ
  rtabmap_util::PointCloudXYZRGB
  rtabmap_util::PointCloudToDepthImage
  rtabmap_util::ObstaclesDetection
  rtabmap_util::PointCloudAggregator
  rtabmap_util::PointCloudAssembler
```

AntoBrandi, 2023-11-21:
@SteveMacenski I addressed it here #652

SteveMacenski, 2023-11-21:
Merged!

closed 2023-11-21 by SteveMacenski