SteveMacenski/slam_toolbox #653: Determination for known regions

Hao-Li-Bachelorarbeit, 2023-11-22:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- ROS Version
  - ROS2 HUMBLE
- Version or commit hash:
  - 
- Laser unit:
  - 
  
#### Steps to reproduce issue
Hi, I've noticed that the map building area of the slam_toolbox is always too small when the robot newly arrives at a pose, whereas, for example, it is only after a rotation that a more reasonable map is obtained, as shown in the picture:
before spin:
[image]
after spin:
[image]
I'm not sure if this is a bug or a feature, and if it is a feature, is it a resolution issue with the laser scan? But as you can see in the picture, the laser dots on the obstacle are still very dense, and in the area close to the obstacle, it should also be open space at the beginning.

#### Expected behavior

#### Actual behavior

#### Additional information

SteveMacenski, 2023-11-22:
It is definitely related to your sensor's resolution. You need enough unique sensor readings of "this is free space" for each cell to count it. That's why near the robot you get some map showing up because enough individual rays overlap enough of the cells to make them count. When you rotate, you trigger another update (or multiple) whos data then also shows from another viewpoint that its clear so then more of the space appears

Hao-Li-Bachelorarbeit, 2023-11-22:
ok, thank you!

closed 2023-11-22 by Hao-Li-Bachelorarbeit