SteveMacenski/slam_toolbox #660: BoundingBox transformation is broken

oenzan, 2024-01-07:
**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
Source
- ROS Version
ROS2
- Version or commit hash:
 foxy-devel

#### Additional information
I am developing rotate map service. I directly use https://github.com/SteveMacenski/slam_toolbox/blob/de13553d9f92e3773da4b2ff6eebd1977b9337f6/src/merge_maps_kinematic.cpp#L195 . I realized that boundingbox transformation is wrong. The corrected version might look like this:

```
// TRANSFORM BOUNDING BOX POSITIONS
BoundingBox2 bbox = scan->GetBoundingBox();
const Vector2<kt_double> bbox_min_corr =
    applyCorrection(bbox.GetMinimum(), submap_correction);
const Vector2<kt_double> bbox_max_corr =
    applyCorrection(bbox.GetMaximum(), submap_correction);
Vector2<kt_double> bbox_min_right_corr{bbox.GetMaximum().GetX(),bbox.GetMinimum().GetY()};
    bbox_min_right_corr = applyCorrection(bbox_min_right_corr, submap_correction);
Vector2<kt_double> bbox_max_left_corr{bbox.GetMinimum().GetX(),bbox.GetMaximum().GetY()};
    bbox_max_left_corr = applyCorrection(bbox_max_left_corr, submap_correction);
bbox.Add(bbox_min_corr);
bbox.Add(bbox_max_corr);
bbox.Add(bbox_min_right_corr);
bbox.Add(bbox_max_left_corr);
scan->SetBoundingBox(bbox);
```

SteveMacenski, 2024-01-08:
Can you open a PR? That'll make it easier for me to look at the diffs to be able to agree with you (and merge the change) :-)

SteveMacenski, 2024-01-09:
Merged

closed 2024-01-09 by SteveMacenski