SteveMacenski/slam_toolbox #663: Is there a sort of maximum_time_interval?

MoffKalast, 2024-01-14:
Let me explain, I have a mapping setup that has a decently sized minimum_travel_distance and minimum_travel_heading, which is nicely efficient when driving along, but causes an annoying edge case when stopped:

https://github.com/SteveMacenski/slam_toolbox/assets/9977799/e4aca493-5990-4742-99bf-1e4adc753af3

As I understand it, what happens is that the robot stops just short of the update distance and sits there with its amcl pose misaligned until it starts running again (problematic if the reason for stopping is for replanning, which would then use a wrong pose). I could simply decrease the update distance limits, but that seems to lead to unstable localization when driving due to CPU limits and wouldn't really fully solve the problem since it's always possible to stop just short of the set distance.

So I'm wondering is there any kind of option I could set to make it run a **one time** pose update if it hasn't been updated for a max time interval while the robot remains stationary?

These are my params for reference, using async online mapping from the humble apt binary:

```yaml
slam_toolbox:
  ros__parameters:

    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    odom_frame: odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan
    use_map_saver: false
    mode: mapping

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.02
    map_update_interval: 5.0
    resolution: 0.05
    max_laser_range: 12.0
    minimum_time_interval: 0.5
    transform_timeout: 0.2
    tf_buffer_duration: 30.0
    stack_size_to_use: 40000000
    enable_interactive_mode: false

    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.5
    minimum_travel_heading: 0.5
    scan_buffer_size: 10
    scan_buffer_maximum_scan_distance: 10.0
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 1.5
    loop_search_maximum_distance: 3.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true
```

Thanks :)

NhatTanXT3, 2024-01-19:
Your odom -> base_footprint looks unreliable.
As I understand, when the robot stops, map->odom also stops updating. The robot pose (on the global frame) relies only on odom->base_footprint until the next slam_toolbox process runs. 

My suggestion for this case is to improve your odometry to be more reliable in the short term. For example, use different odom sources, do sensor fusion, etc

MoffKalast, 2024-01-19:
Well I think I might've hit a wall once or twice on that run, but outside of that the odometry's as good as I can physically get it with existing hardware, when moving linearly it can basically map by just transforming lidar dots onto an image without any pose correction, but rotational odom will always suck to some degree with diff drive encoders on a narrow track width. That's life, and amcl would've almost certainly corrected it immediately without any problems if it actually ran. Overall it's not a huge problem, just really weird behaviour that I've never seen from other slam stacks.

Sure I could add a gyro or some lidar odometry on top that would add more gratuitous complexity or more sensors that would eat even more of the already limited battery, but at that point I'd just need a mapping_toolbox, not a slam_toolbox because I'd be doing all the localization myself 😄

Anyhow, I guess the roundabout answer is that there isn't an option for it. I'll see if I can make the change and a PR if it ends up improving the situation.

closed 2024-01-19 by MoffKalast