SteveMacenski/slam_toolbox #678: [slam_toolbox]: Failed to compute odom pose

jagauthier, 2024-03-07:
Sorry for posting this here. This is not a bug (I don't think)
I'm using ROS2, on docker containers (debian bookworm) with an RPLIDAR A1.

I've started the LIDAR which published /scan.
I've set up a static transform from base_link to laser.
I've started the laser scanner matcher, which publishes odom transforms

Then I start up the SLAM toolkit like so:
```
ros2 launch slam_toolbox online_sync_launch.py \
odom_frame:=odom \
base_frame:=base_link \
map_frame:=map \
scan_topic:=/scan \
map_update_interval:=1 \
max_laser_rang:=5 \
minimum_travel_distance:=0.1 \
use_scan_matching:=true \
minimum_travel_heading:=1.57 \
do_loop_closing:=true
```
 
I've detailed this here:
https://robotics.stackexchange.com/questions/109756/trouble-with-slam-toolkit/109769#109769

My estimate is that this piece of code is returning false:

```
  bool getOdomPose(karto::Pose2 & karto_pose, const rclcpp::Time & t)
  {
    geometry_msgs::msg::TransformStamped base_ident, odom_pose;
    base_ident.header.stamp = t;
    base_ident.header.frame_id = base_frame_;
    base_ident.transform.rotation.w = 1.0;

    try {
      odom_pose = tf_->transform(base_ident, odom_frame_);
    } catch (tf2::TransformException & e) {
      return false;
    }
```

So I would guess it's something wrong with the odom frame to base.  But I don't know how to fix it.

Thanks for looking,

jagauthier, 2024-03-07:
I solved this. Apparently using the launch file is the wrong way to do this.

closed 2024-03-07 by jagauthier

Bopp1312, 2025-04-23:
I am currently wrestling with this issue, I will try to launch the RPLIDAR scan node not using the launch file
I am using the [SLLIDAR](https://github.com/Slamtec/sllidar_ros2) package 
I have found the TF buffer throws an exception in the same place that @jagauthier suspects but the error is that the tf buffer looks at the scan message timestamp and asks for a tranfsform from lidar_link to base_link but it does not have recent enough data in the buffer to do so, it errors saying it has to extrapolate into the future by some amount of time (~10ms-80ms) depending on how fast your /odom topic is published.

Bopp1312, 2025-04-23:
I will try [this](https://robotics.stackexchange.com/questions/109756/trouble-with-slam-toolkit/109769#109769:~:text=Solved.,p%20do_loop_closing%3A%3Dtrue) tonight and hopefully it works, thanks @jagauthier for referencing the solution