SteveMacenski/slam_toolbox #679: Is there a way to output the final poses optimization result?

WANGYINGYU, 2024-03-08:
#### Feature description

I can output poses using /tf in real-time, but I want to use final optimized poses to evaluate the performance of global localization. Is there a way to export this data?

SteveMacenski, 2024-03-08:
Not currently, but I think that could be populated by the occupancy grid generation process, since that's iterating over the sensors at poses to generate the occupancy grid. We could populate a vector of those poses to publish out pretty easily!

WANGYINGYU, 2024-03-11:
> Not currently, but I think that could be populated by the occupancy grid generation process, since that's iterating over the sensors at poses to generate the occupancy grid. We could populate a vector of those poses to publish out pretty easily!

Thanks for your reply. Can you help me find in which file or more specifically in which function the occupancy grid map generation is defined?