SteveMacenski/slam_toolbox #681: Map overlapping problem

anasderkaoui, 2024-03-12:
Hello !
-> My setup:
- YDLIDAR G4
- ROS 2 Humble and Foxy
- Ubuntu 22.04 and 20.04

-> The problem:

I can successfully generate a map, but when I move the robot or when odometry data changes, I get a new scan of the map but overlapping the previous one ! (Fixed Frame is set to "map" and sometimes to "base_footprint" but still got the same behavior)
I want the map to stay put in place (static, not moving with the robot) while the robot is moving to scan the reaminning blank spaces.
I have also tried this with gmapping and got the same problem. Is there a parameter to change in order to fix that ?

Thanks in advance!

CraftyCranberry, 2024-03-14:
Could be some issue with your tf. Triple check the laser to base link tf in your urdf.

anasderkaoui, 2024-03-14:
I am not using a URDF, only tf from the drivers' packages. I checked running: `ros2 run tf2_tools view_frames` and I got the required tf tree: `map -> odom -> base_link -> lidar_frame`. 
I think that if the tf was not set properly, the SLAM algorithm wouldn't even wanna start !
Also on RVIZ, when I add the TF display, I get "Status Ok" with all the Transforms mentionned above!
From some other sources, I found that this maybe due to the LiDAR moving very quickly and jerking. I think this may be the case, I still have to test that!

CraftyCranberry, 2024-03-15:
Nah I mean the tf needs to be correct. An incorrect TF can cause this as far as I know since I had a very similar issue. An Incorrect tf would also look completely normal on the tree.

anasderkaoui, 2024-03-18:
Okay, thank you. How do I know if my tf is correct then ?

PengzhaoWu, 2025-08-14:
@anasderkaoui Hi, I encountered the same problem when I try to use Spot to do SLAM. Have you already solved your problem?

anasderkaoui, 2025-08-16:
I think that the issue was that the LiDAR was not well-placed and the base that it was placed on was **vibrating**. So, I think that might explain why I was getting that issue. Also, the robot does some **very tight turns** that contribute to the overlapping problem !

shourya-886, 2026-08-18:
I had the same issue... my old map and new map kept overlapping... after much debugging, I found that the mapping of my wheels in my code, which interfaces between the Arduino and my ARM, was reversed...

### Setup:
3D-printed chassis
My ARM computer on my robot is the Nvidia Jetson Orin Nano running JetPack 5, Ubuntu 22.04 LTS, ROS2 Humble.
My PC is also running Ubuntu 22.04 LTS and ROS2 Humble...
I am using the Jetson as the computer for ROS2, and it will send commands via UART Communication to myArduino Megaa 2560(which is connected to my motors and L298N motor driver)
I have the RPLIDAR A1M8
U can view my full code at https://github.com/shourya-886/robot_ws

### P.S:
Visit https://github.com/shourya-886/robot_ws/commit/7f2905cfa53c75baf0f277a634dbfcb238914b89 for the changes I made, and the whole interface's code is at: https://github.com/shourya-886/robot_ws/blob/main/src/bumperbot_firmware/src/bumperbot_interface.cpp