SteveMacenski/slam_toolbox #683: Scan data does not update the point behind an obstacle.

Zarqu0n, 2024-04-01:
**Required Info:**

- Operating System:
Ubuntu 22.04
- Installation type:
binary
- ROS Version
ROS2 Humble
- Version or commit hash:
release
- Laser unit:
RP Lidar
  

#### Expected behavior
[image]

#### Actual behavior
[image]

#### Additional information
While mapping with my robot, pushing it causes my legs to block part of the lidar sensor. This results in my footprints appearing on the map and the map being saved in this way. To solve this problem, I added the 'min_laser_range' parameter. However, this solution limits the lidar's detection area in narrow spaces. Cartographer adds lidar data to the map using the 'probability_grid' algorithm. In Slam_toolbox, lidar data is held in a certain buffer before being processed and added to the map. However, since the lidar cannot see the area behind my legs, the points there are not updated. I have tuned the parameters but haven't found one that solves this issue. What solution would you suggest in this situation?
Thanks in advance!

mich1342, 2024-07-14:
Hi @Zarqu0n,

I would recommend you to use laser_filters instead of the `min_laser_range` parameter for this cases. It could be seen [here](https://github.com/ros-perception/laser_filters). I think you could just kind of "turn off" the lidar angle that might read your legs by setting it into either inf or null so that slam_toolbox will just ignore the area scanning your legs.

Zarqu0n, 2024-07-18:
Hello @mich1342  
sorry for replying a little late. Yes, using a laser filter is a more practical solution, but adding an extra node just for this purpose will cause a slight delay in the system. I think it seems like a faster solution to integrate min_laser_range and use it in the slam toolbox just for this task. Also, I don't know if I understood correctly, but let's assume that I turn off the part that sees the legs of the laser during mapping and turn it back on during localization. Will it cause a problem because the pose_graphs in mapping and the pose graphs during localization do not match exactly? Or would it be better?

mich1342, 2024-07-21:
Hi @Zarqu0n ,

For the delay itself I think you could play a little bit with the transform tolerance and I have not found any issue with the slight delay. If you would like to have no delay then I think some lidars driver support to turn off scan partially. Or if you would like to do it the creative way, i suggest you to block the lidar at certain angle really close so that by default it reads below the min scan range. 

For the localization it is not a problem at all