SteveMacenski/slam_toolbox #685: Ubuntu Noble compilation fails: `fatal error: ceres/local_parameterization.h: No such file or directory`

Timple, 2024-04-16:
**Required Info:**

- Operating System:
  - Ubuntu Noble
- Installation type:
  - source
- ROS Version
  - rolling
- Version or commit hash:
  - `ros2` branch
- Laser unit:
  - None
  
#### Steps to reproduce issue
Compile on ubuntu Noble

#### Expected behavior
I would think these kind of templates are mostly self explanatory :smile: 

#### Actual behavior
`fatal error: ceres/local_parameterization.h: No such file or directory`

#### Additional information

@james-ward already solved the compilation it seems: https://github.com/SteveMacenski/slam_toolbox/compare/ros2...james-ward:slam_toolbox:manifold
So credits to him. I didn't verify or validate the output. But I thought I'd create this issue anyhow in case more people run into this.

@james-ward can you perhaps open a PR if you verified operation?

james-ward, 2024-04-16:
@Timple Yep, I was going to open a PR as soon as I could verify it works. At the moment all I know is that it compiles - haven't actually been able to run it on a live system yet (lots of missing packages in noble-rolling still so I can't bring our platform up).

Timple, 2024-04-17:
Awesome, didn't mean to be pushy. Just trying to prevent a lot of duplicate work 🙂

SteveMacenski, 2024-04-21:
Does https://github.com/SteveMacenski/slam_toolbox/pull/687 fully address?

Timple, 2024-04-23:
It does for me, as I only ran into compilation issues for now. But that PR did do some verification :+1:

closed 2024-06-05 by SteveMacenski