SteveMacenski/slam_toolbox #689: Bad obstacle mapping in the map limits

joanvallve, 2024-04-17:
**Required Info:**

- Operating System: Ubuntu 22.04
- Installation type: binaries
- ROS Version: Humble
- Version or commit hash: 2.6.8-1jammy.20240217.104951
- Laser unit: RPLIDAR A1M8 (turtlebot4 simulated)
  
#### Steps to reproduce issue
Launch turtlebot4 simulation in an environment with orthogonal walls starting with zero orientation.

#### Expected behavior
Map the walls on the map limits properly, like all the other walls.

#### Actual behavior
It seems that the walls close to the map limits are not properly mapped and often are mapped as free area.

[image]

#### Additional information
When simulating the robot starting with a different orientation, the walls are properly mapped. To me it seems like a problem of ray casting related with the lack of unknown cells behind. It would be easily fixed adding some extra rows and cols when resizing the map.

[image]

SteveMacenski, 2024-04-20:
How is that map generated? The flat stop of the map really isn’t how lidar data works for the map to have an abrupt end like that. Is this in typical slam operation mode or are you localizing on something else? That map also looks really suspiciously perfect like it’s autogenerated and not from slam

joanvallve, 2024-04-22:
Yes, it is in SLAM with a simulated turtlebot4 initialized with yaw=0.0, not autogenerated or loaded.

SteveMacenski, 2024-04-22:
Please provide something reproducible

joanvallve, 2024-04-23:
Sorry, do you mean a video or instructions to reproduce the error?

SteveMacenski, 2024-04-23:
Instructions, maps, simulation files, etc so I can easily reproduce this myself to look into it. Probably a new github repo for it with instructions in the readme about how to launch and test so I can take a look

joanvallve, 2024-04-24:
Is it enough if I record a rosbag or do you want to simulate?

joanvallve, 2024-04-24:
Ok so I added a readme to the package I use for this simulation: [upc_mrn](https://gitlab.com/joanvallve/upc_mrn)

There you have the dependencies and an example of how to launch the simulation of the turtlebot4 with slam_toolbox. The pictures I uploaded were taken simulating with world `small` and with default starting pose or changing the initial orientation, respectively.

SteveMacenski, 2024-05-03:
Does the robot localize well when in this situation (i.e. doesn't get lost)? What I expect is happening here is not a bug in the core SLAM / sensor matching but in the occupancy grid generation which happens periodically to serve to the user (which isn't used internally). 

If the robot doesn't get delocalized or anything, then that is the most likely situation and that happens here: https://github.com/SteveMacenski/slam_toolbox/blob/ros2/lib/karto_sdk/include/karto_sdk/Karto.h#L5893

Given what you see is on the right side of the map, that is at the `width` maximum value, so there may be an off by 1 error in some validation check if on the grid, we stop raycasting a little early, or round down some bounding box dimension one unit shy. Its hard for me to say which, but let me know if anything in that object or the `Grid` object sticks out to you.

yknighthy, 2024-12-06:
When building the occupancy grid map, SlamToolBox performs raytracing by first finding the coordination of end points in the grid map. At the 4502 line in Karto.h, the function `WorldToGrid` does this work by directly returning the rounded value. 
```
inline Vector2<kt_int32s> WorldToGrid(const Vector2<kt_double>& rWorld, kt_bool flipY = false) const
{
  kt_double gridX = (rWorld.GetX() - m_Offset.GetX()) * m_Scale;
  kt_double gridY = 0.0;

  if (flipY == false)
  {
    gridY = (rWorld.GetY() - m_Offset.GetY()) * m_Scale;
  }
  else
  {
    gridY = (m_Size.GetHeight() / m_Scale - rWorld.GetY() + m_Offset.GetY()) * m_Scale;
  }
  return Vector2<kt_int32s>(static_cast<kt_int32s>(math::Round(gridX)), static_cast<kt_int32s>(math::Round(gridY)));
}
```
However, the above method might push the mapped cell beyond the map boundary if the scan point is currently at the margin. 
By judging whether the result of rounding exceeds the current map before returning the coordination, the obstacle missing problem seems to be fixed for me.
```
inline Vector2<kt_int32s> WorldToGrid(const Vector2<kt_double>& rWorld, kt_bool flipY = false) const
{
  kt_double gridX = (rWorld.GetX() - m_Offset.GetX()) * m_Scale;
  kt_double gridY = 0.0;

  if (flipY == false)
  {
    gridY = (rWorld.GetY() - m_Offset.GetY()) * m_Scale;
  }
  else
  {
    gridY = (m_Size.GetHeight() / m_Scale - rWorld.GetY() + m_Offset.GetY()) * m_Scale;
  }

  kt_int32s gridx = static_cast<kt_int32s>(math::Round(gridX));
  kt_int32s gridy = static_cast<kt_int32s>(math::Round(gridY));
  gridx = (gridx == m_Size.GetWidth()) ? gridx - 1 : gridx;
  gridy = (gridy == m_Size.GetHeight()) ? gridy - 1 : gridy;

  return Vector2<kt_int32s>(gridx, gridy);
}
```

SteveMacenski, 2024-12-06:
Please test that a few times to make sure it fully resolves, but that seems sensible to me on first glance and easy enough for me to merge and backport to all the current distributions :-)

yknighthy, 2024-12-06:
您给我发的邮件已收到。

closed 2026-11-02 by SteveMacenski