SteveMacenski/slam_toolbox #694: UAF bug caused by `laser_callback()`

GoesM, 2024-04-27:
**Required Info:**

- Operating System:
  -  Ubuntu 22.04
- Installation type:
  -  source
- ROS Version
  -  ROS2-humble
- Version or commit hash:
  -  the latest
- Laser unit:
  -  defaulted
  
#### Steps to reproduce issue

I use slam-toolbox (async) by following command :
```sh
#!/bin/bash
export ASAN_OPTIONS=halt_on_error=0:new_delete_type_mismatch=0:detect_leaks=0:log_path=asan
source install/setup.bash
ros2 launch slam_toolbox online_async_launch.py
```
Running Slam-Toolbox within AddressSanitizer , I always faced to such UAF report  during shutdown-period

#### Expected behavior
No UAF occurs

#### Actual behavior

we could always face to an ASAN-report about UAF bug as following:
```
=================================================================
==18319==ERROR: AddressSanitizer: heap-use-after-free on address 0x6060000736b8 at pc 0x7605a1e5f0d5 bp 0x76058f4c8340 sp 0x76058f4c8338
READ of size 8 at 0x6060000736b8 thread T20
    #0 0x7605a1e5f0d4 in std::_Rb_tree<karto::Name, std::pair<karto::Name const, karto::Sensor*>, std::_Select1st<std::pair<karto::Name const, karto::Sensor*> >, std::less<karto::Name>, std::allocator<std::pair<karto::Name const, karto::Sensor*> > >::find(karto::Name const&) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x65f0d4) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #1 0x7605a1e5ec16 in karto::SensorManager::GetSensorByName(karto::Name const&) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x65ec16) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #2 0x76059d6a9d28 in karto::GridIndexLookup<unsigned char>::ComputeOffsets(karto::LocalizedRangeScan*, double, double, double) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libkartoSlamToolbox.so+0x4a9d28) (BuildId: 438116af7e8121c8d8c716f02dee888c7979effe)
    #3 0x76059d65afb3 in karto::ScanMatcher::CorrelateScan(karto::LocalizedRangeScan*, karto::Pose2 const&, karto::Vector2<double> const&, karto::Vector2<double> const&, double, double, bool, karto::Pose2&, karto::Matrix3&, bool) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libkartoSlamToolbox.so+0x45afb3) (BuildId: 438116af7e8121c8d8c716f02dee888c7979effe)
    #4 0x76059d6ad3bf in double karto::ScanMatcher::MatchScan<std::vector<karto::LocalizedRangeScan*, std::allocator<karto::LocalizedRangeScan*> > >(karto::LocalizedRangeScan*, std::vector<karto::LocalizedRangeScan*, std::allocator<karto::LocalizedRangeScan*> > const&, karto::Pose2&, karto::Matrix3&, bool, bool) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libkartoSlamToolbox.so+0x4ad3bf) (BuildId: 438116af7e8121c8d8c716f02dee888c7979effe)
    #5 0x76059d6776c7 in karto::Mapper::Process(karto::LocalizedRangeScan*, karto::Matrix3*) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libkartoSlamToolbox.so+0x4776c7) (BuildId: 438116af7e8121c8d8c716f02dee888c7979effe)
    #6 0x7605a1dfa61a in slam_toolbox::SlamToolbox::addScan(karto::LaserRangeFinder*, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, karto::Pose2&) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x5fa61a) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #7 0x7605a2ab0c89 in slam_toolbox::AsynchronousSlamToolbox::laserCallback(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libasync_slam_toolbox.so+0xb3c89) (BuildId: ab3d6662343414a1ceb1f167a59faeabcec8089b)
    #8 0x7605a200c7b1 in void std::__invoke_impl<void, void (slam_toolbox::SlamToolbox::*&)(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>), slam_toolbox::SlamToolbox*&, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&>(std::__invoke_memfun_deref, void (slam_toolbox::SlamToolbox::*&)(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>), slam_toolbox::SlamToolbox*&, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x80c7b1) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #9 0x7605a200bed2 in message_filters::CallbackHelper1T<std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call(message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, bool) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x80bed2) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #10 0x7605a1fa434b in message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call(message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x7a434b) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #11 0x7605a2003cba in tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::transformReadyCallback(tf2_ros::TransformStampedFuture const&, unsigned long) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x803cba) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #12 0x7605a20083ce in std::_Function_handler<void (tf2_ros::TransformStampedFuture const&), std::_Bind<void (tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::* (tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>*, std::_Placeholder<1>, unsigned long))(tf2_ros::TransformStampedFuture const&, unsigned long)> >::_M_invoke(std::_Any_data const&, tf2_ros::TransformStampedFuture const&) (/home/****/slam_toolbox_humble_fork/install/slam_toolbox/lib/libtoolbox_common.so+0x8083ce) (BuildId: a965c84ea396d1a224891c94c234cfed672c2bd1)
    #13 0x7605a2953e8d  (/opt/ros/humble/lib/libtf2_ros.so+0x4fe8d) (BuildId: dceacb25e05c8a82678784802b23fb16ba98d172)
    #14 0x76059f94a65e in tf2::BufferCore::testTransformableRequests() (/opt/ros/humble/lib/libtf2.so+0x1065e) (BuildId: 5677f8e557cfe0980662adcc0a17f03987f8b7f1)
    #15 0x76059f94d3fa in tf2::BufferCore::setTransformImpl(tf2::Transform const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (/opt/ros/humble/lib/libtf2.so+0x133fa) (BuildId: 5677f8e557cfe0980662adcc0a17f03987f8b7f1)
    #16 0x76059f94db29 in tf2::BufferCore::setTransform(geometry_msgs::msg::TransformStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (/opt/ros/humble/lib/libtf2.so+0x13b29) (BuildId: 5677f8e557cfe0980662adcc0a17f03987f8b7f1)
    #17 0x7605a295cbb0 in tf2_ros::TransformListener::subscription_callback(std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>, bool) (/opt/ros/humble/lib/libtf2_ros.so+0x58bb0) (BuildId: dceacb25e05c8a82678784802b23fb16ba98d172)
    #18 0x7605a159f1c9 in std::_Function_handler<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>), std::_Bind<void (tf2_ros::TransformListener::* (tf2_ros::TransformListener*, std::_Placeholder<1>, bool))(std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>, bool)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>&&) (/opt/ros/humble/lib/librviz_default_plugins.so+0x79f1c9) (BuildId: 9da6509995fc2c916736fdd2d61b58739854b36d)
    #19 0x7605a15ae031 in std::__detail::__variant::__gen_vtable_impl<std::__detail::__variant::_Multi_array<std::__detail::__variant::__deduce_visit_result<void> (*)(rclcpp::AnySubscriptionCallback<tf2_msgs::msg::TFMessage_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > >, rclcpp::MessageInfo const&)::'lambda'(auto&&)&&, std::variant<std::function<void (tf2_msgs::msg::TFMessage_<std::allocator<void> > const&)>, std::function<void (tf2_msgs::msg::TFMessage_<std [... cut ...]
the function `laserCallback()` is bind to the `scan_filter_sub_` and `scan_filter_` as following:

https://github.com/SteveMacenski/slam_toolbox/blob/94cec982a7f850818187c81295d1212f145efe37/src/slam_toolbox_common.cpp#L233-L241

but here's no `reset()` for such two pointer in the destructor

https://github.com/SteveMacenski/slam_toolbox/blob/94cec982a7f850818187c81295d1212f145efe37/src/slam_toolbox_common.cpp#L85-L101

So that, the callback-function might be still working after the node is destructed and cause the UAF bug as a result.

GoesM, 2024-04-27:
well, it seems that such issue has been fixed in `ros2` branch :

https://github.com/SteveMacenski/slam_toolbox/blob/a3442d2f6824ff058fab0cb0a635e7a454294855/src/slam_toolbox_common.cpp#L240-L260

but no such code to reset these pointers in both `iron` and `humble` branch.

GoesM, 2024-06-28:
Is it neccessary for me to use `cherry-pick` ? 🤔 I think the past commit for this issue would change `humble` a lot:

https://github.com/SteveMacenski/slam_toolbox/commit/de13553d9f92e3773da4b2ff6eebd1977b9337f6

SteveMacenski, 2024-06-28:
I'd try cherry-pick, but if it looks too hard, I understand manual for this case.