SteveMacenski/slam_toolbox #705: Detecting Smaller Object

lesu0330, 2024-06-13:
#### Feature description

Create a way so that the slam can map smaller objects. This will be able to solve problems when needing to use slam for smaller obstacles. Currently the slam toolbox does not consider 

#### Implementation considerations

I do not know how this feature can be implemented completely but maybe have additional parameters in the param file or be able to adjust the ceres solver.

SteveMacenski, 2024-06-13:
What do you mean "smaller objects"? Are these objects that your lidar is not picking up? That would seem more a concern with lidar resolution w.r.t. the obstacles you want to map issue than the algorithm itself. But please clarify :-)

SteveMacenski, 2024-06-13:
No, there is no way to do that with the SLAM problem. The better and more common solution is to have a detector _before_ SLAM which is detecting the small obstacle in your sensor reading and then publishing sensor data with that obstacle removed. Then that topic can be used by SLAM and it never knew about the small obstacle in the first place. These are pretty common to have sensor filtering and processing pipelines for depth/lidar measurements before ever hitting the SLAM or obstacle avoidance algorithms.

I think that's the right/best solution!

closed 2024-06-13 by SteveMacenski

lesu0330, 2024-06-13:
Just to clarify what you have said. Are you indicating that the only way to detect smaller object or ignoring an object is to change the lidar measurements/performance. Make it worse if I wish to ignore the obstacles and make it better if I do want to detect it? Then may I also ask what the tolerance and trust areas do in the ceres_solver.cpp?

SteveMacenski, 2024-06-13:
`lidar -> pre processing stages -> {slam, perception, obstacle avoidance, etc}` is the common pipeline. If you want to ignore some data based on size or detected status, you should have a preprocessing stage for that.

lesu0330, 2024-06-13:
So there will need to be a preprocessing stage before SLAM? That is not part of the slam toolbox repository?