SteveMacenski/slam_toolbox #706: Lifelong Mapping not Found in Humble branch,  ros2 branch could not be compiled in Humble

mich1342, 2024-06-14:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - humble binaries and source
- ROS Version
  - ROS2 Humble
- Version or commit hash:
  - ros2 & humble
- Laser unit:
  - RPLidar S3
  
#### Steps to reproduce issue
Scenario A:
1. Install slam toolbox from binary release `sudo apt-get install ros-humble-slam-toolbox`

Scenario B:
1. Install slam toolbox from source branch humble `git clone -b humble https://github.com/SteveMacenski/slam_toolbox.git`

Scenario C:
1. Install slam toolbox from source branch ros2 `git clone -b ros2 https://github.com/SteveMacenski/slam_toolbox.git`

#### Expected behavior
Lifelong mapping should works

#### Actual behavior
Scenario A:
`lifelong_launch.py` not found

Scenario B:
`lifelong_launch.py` not found

Scenario C:
got `fatal error: ceres/autodiff_manifold.h: No such file or directory`

#### Additional information

mich1342, 2024-06-14:
It turns out I just need to run the `online_async_launch` and provide posegraph map (.data and .posegraph) instead of .yaml and .pgm

closed 2024-06-14 by mich1342

tanmayrspace, 2024-06-29:
Hi @mich1342  were able to solve this problem
 fatal error: ceres/autodiff_manifold.h: No such file or directory
 i am facing the same issue while building from source

SteveMacenski, 2024-06-29:
Use the correct branch for your distribution. ros2 targets rolling as is standard convention. Rolling is on 24.04 where there were API updates to Ceres 🙂

tanmayrspace, 2024-07-01:
Hi @SteveMacenski Thanks For replying.  Basically i wanted to use the lifecycle node implementation of the slam_toolbox that was on the current ros2 branch, as you mentioned this is only supported on 24.04. so I eventually used the old implementation of the ceres-solver from the humble branch and lifecycle implementation from the current branch. As of now it is not creating any problems.

aravindsairam, 2024-12-08:
Hello @tanmayrspace, can you explain the changes in detail you made that to work. Im facing the same issue. I also want to the lifecycle manager node implementation for my ros2 humble package. It could be very helpful. Thanks !!!