SteveMacenski/slam_toolbox #707: Set minimum Laser Range for SLAM

mich1342, 2024-06-14:
#### Feature description

I am thinking of adding additional parameters to set the minimum laser range (as it now support the `max_laser_range`). I have a use cases where my robot goes under a trolley and lurk / pull the trolley around. I need to do lifelong mapping due to the highly dynamic environment. I got a problem where my laserscan data read the trolley legs and add it into the map causing the robot could not move anywhere.

#### Implementation considerations

It could be solved either by:
1. Add `min_laser_range` parameters for slam_toolbox to ignore laserscan readings that are lower than the value
2. Filter my laserscan data out manualy 

I personally prefer the first option since I might need the second one for my navigation just in case I have rectangle instead of square.

SteveMacenski, 2024-06-14:
It will actually use the laser scan message's minimum range if you set that properly to the minimum you'd like to use https://github.com/SteveMacenski/slam_toolbox/blob/8749e3392f80e400409da00a9e7fc401ba0818d5/src/laser_utils.cpp#L103

But, if you want to keep that the same for other applications, having the override for minimum here like we do maximum seems reasonable!

mich1342, 2024-06-25:
Hi @SteveMacenski ,

Thank you for your reply. In my applications, using the laser scan message's minimum range is not really applicable since it will has different requirement of minimum range when the AMR is not lurking / pulling trolley. In that cases I would like to have smaller minimum ranges. It would be great if we have minimum range like the maximum without the needs of changing the laser scan mesage's minimum range.

mich1342, 2024-06-26:
Hi @SteveMacenski ,

I have created a PR for this feature and have tested it using both simulation and real robot. Could you help to verify whether if the code I implemented align with what did you mean previously. Thank you

SteveMacenski, 2024-06-26:
OK! I gave it a review, but need some un-formatting of that file to do a meaningful review first since the diff is just _the entire file_

SteveMacenski, 2024-06-28:
Merged

closed 2024-06-28 by SteveMacenski