SteveMacenski/slam_toolbox #720: Message Filter dropping message: frame 'laser' at time 1721657901.229 for reason 'discarding message because the queue is full'

urwa95, 2024-07-22:
[sync_slam_toolbox_node-2] [INFO] [1721657901.312031566] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.150 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-2] [INFO] [1721657901.394929913] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.229 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-2] [INFO] [1721657901.477027034] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.312 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-2] [INFO] [1721657901.560229949] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.395 for reason 'discarding message because the queue is full'
-- cannot get rid of this error, tried changing queue size and launch file many times

Raunakabhani, 2024-09-25:
check if topic /odom is getting any data or not, you can use laser odom by using package rf2o_laser_odometry

FLYivan, 2024-11-13:
> [sync_slam_toolbox_node-2] [INFO] [1721657901.312031566] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.150 for reason 'discarding message because the queue is [...]

Actually, I encountered the same problem, but previous responses from others inspired me and I ultimately solved the problem. Below, I will summarize my experience
This situation is a problem with the TF relationship
1、check your TF tree , Ensure the integrity of the structure from map to Odom and then to base_ink, and check it through "rqt_tf_tree"
2、Most overlooked situations，check the configuration files in the config folder of the slam toolbox，because the frame_id in the yaml file may different from your customized frame_id (for example odom、 base、 map），if it is different，modify the yaml  file

maxxbolotov, 2024-12-02:
> Actually, I encountered the same problem, but previous responses from others inspired me and I ultimately solved the problem. Below, I will summarize my experience This situation is a problem with [...]

I faced same issue but quiet don't understand you second advise. Can you please provide what and where should I change to solve the problem?

SteveMacenski, 2024-12-02:
Closing - as this is a user configuration issue as folks have pointed out. Please keep chatting / helping here as needed, but closing the issue since it was resolved.

closed 2024-12-02 by SteveMacenski

maxxbolotov, 2024-12-02:
> > > [sync_slam_toolbox_node-2] [INFO] [1721657901.312031566] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.150 for reason 'discarding message because the queue is full' [sync_slam_toolbox_node-2] [INFO] [1721657901.394929913] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.229 for reason 'discarding message because the queue is full' [sync_slam_toolbox_node-2] [INFO] [1721657901.477027034] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.312 for reason 'discarding message because the queue is full' [sync_slam_toolbox_node-2] [INFO] [1721657901.560229949] [slam_toolbox]: Message Filter dropping message: frame 'laser' at time 1721657901.395 for reason 'discarding message because the queue is full' -- cannot get rid of this error, tried changing queue size and launch file many times
> > 
> > 
> > Actually, I encountered the same problem, but previous responses from others inspired me and I ultimately solved the problem. Below, I will summarize my experience This situation is a problem with the TF relationship 1、check your TF tree , Ensure the integrity of the structure from map to Odom and then to base_ink, and check it through "rqt_tf_tree" 2、Most overlooked situations，check the configuration files in the config folder of the slam toolbox，because the frame_id in the yaml file may different from your customized frame_id (for example odom、 base、 map），if it is different，modify the yaml file
> 
> I faced same issue but quiet don't understand your second advise. Can you please provide what and where should I change to solve the problem?