SteveMacenski/slam_toolbox #722: Extracting exact cell occupancy alongside cell state result.

Paskul, 2024-07-29:
#### Feature description

#### Implementation considerations

SteveMacenski, 2024-07-29:
We do not currently expose this information in any externally consumable output. That would need to have changes to the software so that this information can be extracted and published when generating the occupancy grid in the update map thread. 

> I saw a previous PR where covariance was added as a message in https://github.com/SteveMacenski/slam_toolbox/issues/427 leading to https://github.com/SteveMacenski/slam_toolbox/pull/508, though I am still confused as to how I could implement this for my own needs.

I don't think either of those help you in what you're looking for.

Paskul, 2024-07-29:
Thanks, I see. In making the code changes then, I see the OccupancyGrid with CellUpdater and I feel as if I can make immediate changes to expose hitRatio to the OccupancyGrid class, but what suggestions might you have to put this through to a message/topic? Where/how is the occupancy grid extracted to msgs in this repo?

SteveMacenski, 2024-07-30:
I don't know how you'd like to consume that hit vs miss data, but something along the lines of the structure of the occupancy grid is probably good