SteveMacenski/slam_toolbox #723: Motion compensation to undistort raw LIDAR readings

billkotsias, 2024-07-30:
#### Feature description
When robot is moving during mapping, LIDAR readings produce a distorted view of its surroundings. AFAICS, slam_toolbox compensates for all scan poses but doesn't do motion compensation within a LIDAR scan.

#### Implementation considerations
There are already motion compensation filters, however they produce a `PointCloud`. The [default scan message](https://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/LaserScan.html) that **slam_toolbox** expects is very limited, as it can only represent rays that have the same angle between them, which is **not** the case when doing motion compensation.

IMHO it would be very robust to be able to feed **slam_toolbox** with a pre-compensated LIDAR PointCloud instead of `LaserScan`.

billkotsias, 2024-07-30:
[image]
- red distorted
- white compensated

SteveMacenski, 2024-07-30:
You can feed in your undistorted scan into slam toolbox using a scan undistortion node between the raw sensor data and the slam method

billkotsias, 2024-07-31:
@SteveMacenski I'm a newbie to SLMT... Could you please point me to some guide/code on how to insert such a node between raw data and SLAM method?

Is there a point in code where raw LIDAR rays are converted to 2D points, so that I can add my filter there?

SteveMacenski, 2024-07-31:
1. In your launch file, remap the input topic to the motion compensator to your sensor & output topic to some intermediate topic X
2. In your launch file, remap the input topic for slam toolbox to X (or use the provided parameter in the parameter file)
3. It works!

closed 2024-07-31 by SteveMacenski

billkotsias, 2024-08-01:
Thank you for taking the time!
I'm afraid it won't be that simple though because - like I said in original post :
> There are already motion compensation filters, however they produce a **_PointCloud_**. The [default scan message](https://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/LaserScan.html) that [...]

So I guess I would need to adapt the SLMT lidar message interpreter.