SteveMacenski/slam_toolbox #725: Troubleshoot weird looking map

pitosalas, 2024-08-08:
My setup:

ROS1, Ubuntu 20.04, Noetic

I am experimenting for the first time with slam_toolbox. It is creating an odd looking map and I thought you could eyeball it and see what might be off. I just drove it around my basement for about 3 minutes. Nothing was moving. I believe my odometer and. scan are correct. Thanks for any insights!

[image]

I can provide a lot more info, but as a start, I thought others could spot the problem.

SteveMacenski, 2024-08-08:
Ask on robotics stack exchange - we don't provide configuration help here beyond the documentation on each parameter, roscon videos, and tutorials that exist

closed 2024-08-08 by SteveMacenski

pitosalas, 2024-08-08:
Understood. Thanks for your fine work on this!