SteveMacenski/slam_toolbox #728: SLAM map update interval is fixed as 10.0s

JYKimm, 2024-09-11:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binary
- ROS Version
  - Humble
- Version or commit hash:
  - 
- Laser unit:
  - SICK nano3
  
#### Steps to reproduce issue
I gave a namespace to my mobile robot as robot1.
to do that, I added remapping and namespace parameters to online_sync_launch.py(and async too), and my robot's robot_bringup_launch (including robot_state_publisher)
as a result, topics for SLAM got to have namespace like this,
/tf->/robot1/tf ... same to /odom, /scan....  etc

But, after that SLAM works well but /map's publishing interval was fixed as 10s (0.1Hz)
I checked map_update_interval in mapper_params_online_sync.yaml and it was 5.0
I tried to change the number to 3.0, 15.0, 20.0... but always /map update interval was 10 second.
it was all same for online_async_launch, online_sync_launch

#### Expected behavior
/map topic update interval should be map_update_interval in mapper_params_xxxx.yaml

#### Actual behavior
/map update interval is fixed 10 second if namespace is given in slam toolbox launcher.

#### Additional information

def generate_launch_description():
    use_sim_time = LaunchConfiguration('use_sim_time')
    slam_params_file = LaunchConfiguration('slam_params_file')

    declare_use_sim_time_argument = DeclareLaunchArgument(
        'use_sim_time',
        default_value='true',
        description='Use simulation/Gazebo clock')
    declare_slam_params_file_cmd = DeclareLaunchArgument(
        'slam_params_file',
        default_value=os.path.join(get_package_share_directory("slam_toolbox"),
                                   'config', 'mapper_params_online_async.yaml'),
        description='Full path to the ROS2 parameters file to use for the slam_toolbox node')

    start_async_slam_toolbox_node = Node(
        parameters=[
          slam_params_file,
          {'use_sim_time': use_sim_time}
        ],
        package='slam_toolbox',
        executable='async_slam_toolbox_node',
        name='slam_toolbox',
        output='screen',
        **namespace='robot1',
        remappings=[('/tf','tf'),('/tf_static','tf_static'),('/scan','scan'),('/odom','odom')]**
        )

    ld = LaunchDescription()

    ld.add_action(declare_use_sim_time_argument)
    ld.add_action(declare_slam_params_file_cmd)
    ld.add_action(start_async_slam_toolbox_node)

    return ld

SteveMacenski, 2024-09-11:
> I gave a namespace to my mobile robot as robot1.

I'm guessing this is the issue, you didn't add the namespace to your configuration file so that it gets the parameters. Without that, it just uses the defaults in the code. The namespace of it shouldn't be `slam_toolbox:` (or whatever it is now). It should be `robot1/slam_toolbox:`

JYKimm, 2024-09-12:
Thank you, problem solved

closed 2024-09-12 by JYKimm